rusefi/rusefi_documentation #1: Find a way to convert .md pages from here to .pdf files

rusefi, 2019-01-11:
Some approaches proposed at https://superuser.com/questions/689056/how-can-i-convert-github-flavored-markdown-to-a-pdf

Need to try one or a couple and see how that goes

rusefi, 2019-01-11:
https://github.com/rusefi/rusefi_documentation/tree/master/misc/Saab_Trionic_8_Combustion%20Detection%20Module_on_Mazda_Miata_running_rusEfi could be used a test page for the conversion

command line would be best, maybe we need a complete change of approach? @ZHoob2004 mentioned and I have no idea on pros and cons

rusefillc, 2022-12-12:
Nope :(

mi-hol, 2023-01-19:
I succeeded letting pandoc create a pdf document from START-HERE.md 

[StartHere.pdf](https://github.com/rusefi/rusefi_documentation/files/10461397/StartHere.pdf)

Setup and commands used:
!!note: draft!!
```powershell
#install pandoc
winget install JohnMacFarlane.Pandoc
#install context (pdf-engine)
download http://lmtx.pragma-ade.com/install-lmtx/context-win64.zip
unzip to c:\context
cd c:\context
cmd.exe
.\install.bat
exit
#note: context setenv.bat has a bug and sets PATH incorrectly!
$env:path = "$env:path;C:\ConText\tex\texmf-win64\bin"
pandoc --pdf-engine=context -V pdfa .\START-HERE.md -o .\StartHere.pdf
```

mi-hol, 2023-01-19:
@rusefillc 

Saab-Trionic-8-Combustion-Detection-Module-on-Mazda-Miata-running-rusEFI.md gives a 14MB pdf, some "tuning" required
[Saab-Trionic-8-Combustion-Detection-Module-on-Mazda-Miata-running-rusEFI.pdf](https://github.com/rusefi/rusefi_documentation/files/10461495/Saab-Trionic-8-Combustion-Detection-Module-on-Mazda-Miata-running-rusEFI.pdf)

 pandoc --pdf-engine=context -V pdfa .\Saab-Trionic-8-Combustion-Detection-Module-on-Mazda-Miata-running-rusEFI.md -o .\\Saab-Trionic-8-Combustion-Detection-Module-on-Mazda-Miata-running-rusEFI.pdf

mi-hol, 2023-01-20:
Task  ["PDF software?](https://github.com/rusefi/rusefi/projects/5#card-85202358 )" for Dron is therefore solved from my view

@dron0gus is this maybe your task (user names are different)?

mi-hol, 2023-03-09:
@ElDominio as the initial request might have come from you I wonder if you are still interested in the topic "printed manual"?