rusefi/rusefi_documentation #27: multi-spark manual

rusefi, 2020-04-23:
we have mutli spark with zero content

rusefi, 2020-04-27:
@mck1117 I would like to reiterate the importance of this ticket in my opinion.

mck1117, 2020-04-27:
didn't @SVeilleux9 write a page on this?

rusefi, 2020-04-27:
What is the URL for that page?

OrchardPerformance, 2020-04-28:
I have put a template for the multi-spark into the last PR, @SVeilleux9 has some content to drop there. 
Otherwise I can take that over if required.

SVeilleux9, 2020-04-29:
I have the document, I just need to pull together several TS screenshots and I can upload it.

OrchardPerformance, 2020-05-22:
@SVeilleux9 - The page looks like it has had a lot of updates? 
Is this closed?

rusefillc, 2020-09-12:
@mck1117 @SVeilleux9 ?

SVeilleux9, 2021-01-30:
@OrchardPerformance Just a heads up I wrote that page without knowing much about multi-spark. I understand what it does, but it could probably use some wording as to why you would use it, why you should care.

duhafnusa4, 2021-07-13:
We have https://github.com/rusefi/rusefi/wiki/Multi-Spark

Closing ticket

rusefillc, 2021-07-13:
is feature list article linking at that page?

duhafnusa4, 2021-07-13:
yes

https://github.com/rusefi/rusefi/wiki

closed 2021-07-13 by duhafnusa4