rusefi/rusefi_documentation #35: "download" and boards pages improvements

rusefi, 2020-04-29:
https://github.com/rusefi/rusefi/wiki/Download
still has links to old simulator page
let's remove stuff from legacy wiki simulator page and replace with just a link to new simulator page

We need some small/medium size icons/images for boards and links between "download" and sections of "hardware" page

on hardware page we need a bit more content maybe take something from forum - see links at 
https://github.com/rusefi/rusefi/wiki/Hardware

move more content from https://rusefi.com/wiki/index.php?title=Hardware:Boards_Comparison

https://rusefi.com/build_server/ is really ugly but the benefit of that page is file modification maybe we would have to generate .md page automatically with a script to have better look but last modification date

rusefi, 2020-04-29:
@Timbelom please look into all this

duhafnusa4, 2021-07-22:
@Timbelom this was assigned to you over a year ago. Please respond if you are able to support. If not ticket will be re-assigned.

mi-hol, 2022-12-11:
@duhafnusa4 is below mention issue solved from your view?
"https://github.com/rusefi/rusefi/wiki/Download
still has links to old simulator page
let's remove stuff from legacy wiki simulator page and replace with just a link to new simulator page"