rusefi/rusefi_documentation #36: more "overview" content

rusefi, 2020-04-29:
we need more content on 
https://github.com/rusefi/rusefi/wiki/overview

we shall take parts of https://rusefi.com/wiki/index.php?title=Main_Page at least embed some of youtube 

let's see if we can copy/paste anything from https://rusefi.com/forum/viewtopic.php?f=13&t=1658 presentation

rusefi, 2020-04-29:
@duhafnusa4 can you look into this?

duhafnusa4, 2020-04-29:
I'm on it

duhafnusa4, 2020-04-29:
images for screenshots of video links
[image]

duhafnusa4, 2020-04-29:
updated to look better. We can add more if we want. 

[image]

duhafnusa4, 2020-04-29:
[image]

duhafnusa4, 2021-07-13:
This page now has lots of good information. Closing ticket.

closed 2021-07-13 by duhafnusa4