rusefi/rusefi_documentation #38: dead links - toolset and audit

rusefi, 2020-04-29:
we have a few broken links even image links on the new wiki

we need to document our favorite online or offline took to find broken links

we shall see what's the status with broken link and what can we fix

rusefi, 2020-04-29:
@Timbelom can you google what are helpful tools here?

OrchardPerformance, 2020-06-01:
My favorite offline link detection method is @GlitchKnitter

rusefi, 2020-06-01:
@OrchardPerformance is that Artificial Intelligence or Natural Intelligence System?

chuckwagoncomputing, 2021-02-16:
I have created a tool:
https://github.com/rusefi/rusefi_documentation/blob/master/wiki-tools/brokenlinks.sh

I have created an issue with the remaining links that is has found that cannot be fixed automatically by it:
https://github.com/rusefi/rusefi_documentation/issues/123

closed 2021-02-18 by chuckwagoncomputing