rusefi/rusefi_documentation #39: Update rusEfi map & mention map on wiki

rusefi, 2020-04-29:
jvluntzel now has access to edit rusEfi map

it could be that https://rusefi.com/forum/viewtopic.php?p=31825#p31825 is the last update I did? 

once we update we shall post on the forum "everyone added" for us to know where is the threshold for next time

duhafnusa4, 2021-07-13:
@rusefillc its been a long time since i logged in. Can you remind me how? I will update after.

duhafnusa4, 2021-07-22:
Map has been updated. We should set how often we want to update the list, and I will add that to my calendar to perform. Then we can close this ticket.

chuckwagoncomputing, 2022-02-19:
For some reason I'm not showing up on the map in the first message on the forum. I checked a few other people who joined recently and they're missing too.

duhafnusa4, 2022-06-03:
So..do we really care about the map?

rusefillc, 2022-06-03:
Not really

closed 2022-06-03 by rusefillc