rusefi/rusefi_documentation #40: Add microRusEfi documentation to index page

rusefi, 2020-04-29:
We need a reference to MRE documentation on wiki home.md

MRE documentation itself is a mess with 70% on RE wiki and 30% on microRusEfi github

https://github.com/rusefi/hw_microRusEfi/wiki is the cherry on the cake

rusefi, 2020-04-29:
@duhafnusa4 can you?

duhafnusa4, 2020-04-29:
@rusefi do I need to fork to edit wiki home.md? sorry still learning the ins and outs of GH

duhafnusa4, 2020-04-30:
[image]
[image]

rusefi, 2020-05-30:
@OrchardPerformance close or not close?

duhafnusa4, 2021-07-13:
https://github.com/rusefi/rusefi/wiki/Hardware-microRusEFI is the page for MRE documentation now. Closing ticket.

closed 2021-07-13 by duhafnusa4