rusefi/rusefi_documentation #66: case studies - baseline good tunes with logs showing few important modes

rusefi, 2020-06-03:
we need at least three tunes for three vehicles, with set of logs we are happy with
1) cranking and warm-up
2) idling
3) driving with acceleration

mck1117, 2020-06-04:
I'm happy to record a drive today and share the log

rusefi, 2020-06-04:
drive would be "#3"

also need cold start/warm-up & warm idling :)

mck1117, 2020-06-04:
drive = start logging, start car, leave driveway, go somewhere, shut down, etc.  So it probably includes all 3!

mck1117, 2020-06-04:
https://rusefi.com/forum/viewtopic.php?p=36923#p36923

rusefi, 2020-06-04:
@andreika-git FYI https://rusefi.com/forum/viewtopic.php?p=36923#p36923

duhafnusa4, 2021-07-13:
pick and choose 3 good ones from REO?

@duhafnusa4 
@mck1117 

Who else?

Where do we want to display this information? This ticket and https://github.com/rusefi/rusefi_documentation/issues/151 seem redundant

rusefillc, 2021-07-13:
@duhafnusa4 without matching videos we have _nothing_. we have a lot of data piles together but this ticket is about pointing at the diamonds very specifically.

duhafnusa4, 2021-07-13:
Makes sense. You'll have mine in a few weeks. @mck1117 can you make a quick vid of the 240?

rusefillc, 2021-07-13:
"few important modes" and "quick vid" are two very different things

the goal is not to just close tickets, right? some things are actually asking for labor.

duhafnusa4, 2021-07-22:
I understand @rusefillc . This seems inline with 

https://github.com/rusefi/rusefi_documentation/issues/151

Is this ticket #66 redundant? ie can we combine #66 and #151?