rusefi/rusefi_documentation #76: hints for USB connectivity

rusefillc, 2020-08-15:
Matthew Kennedy  11:35 AM
It needs to be shielded all the way to the connector
Don't connect the shield at the ECU end
There should be exactly one ground from laptop to ECU, and it needs to be in the usb cable, no splices.

Alex Miculescu  11:36 AM
it is there

Andrey  11:37 AM
#dev_documentation this? ^^^ @Simon - Orchardperformance?

Alex Miculescu  11:37 AM
i use the pig tail then the usb cable, but  have a piece that it is not shielded

Matthew Kennedy  11:37 AM
How long is non shielded piece

Alex Miculescu  11:39 AM
unplugged fan, now it is working :)))
11:39
at least is more stabe the connection

OrchardPerformance, 2021-03-06:
Will also link my quick video on this subject to the wiki

duhafnusa4, 2021-07-13:
added information to Q and A section of https://github.com/rusefi/rusefi/wiki/HOWTO-quick-start

closing ticket

closed 2021-07-13 by duhafnusa4