rusefi/rusefi_documentation #100: rusEFI at SF

rusefillc, 2020-09-12:
sync got broken?

https://sourceforge.net/p/rusefi/code/HEAD/tree/trunk/

fix sync? drop SF? else?

chuckwagoncomputing, 2021-02-16:
Is SF providing any value at this point?

mck1117, 2021-02-16:
no

chuckwagoncomputing, 2021-02-16:
burn it

rusefillc, 2021-02-16:
"burn" is vague.

I am not sure what are the relevant buttons if any https://webapps.stackexchange.com/questions/42490/can-you-delete-a-project-on-sourceforge

mck1117, 2021-02-16:
ok well that's pretty dumb, thanks sourceforge

mck1117, 2021-02-16:
maybe you can open an issue on sourceforge itself? https://sourceforge.net/p/forge/site-support/14223/

chuckwagoncomputing, 2021-02-16:
When I search "open source ecu" in DuckDuckGo, the SF repo is the 6th result, and the only other rusEFI result on the first page is the Hackaday article about Frankenstein. That's got to be confusing to someone when they go to the repo and see that it hasn't been updated since 2019.

rusefillc, 2021-02-16:
I wonder if it would be better to drop all files via SVN and keep the "moved to GH" readme?

rusefillc, 2021-04-27:
https://sourceforge.net/p/rusefi/code/HEAD/tree/firmware_archive/ is the most useful part we need to preserve somewhere

duhafnusa4, 2021-07-13:
Looks like link works properly. @mck1117 can you validate? If so please close.

rusefillc, 2021-07-13:
@duhafnusa4 you are scaring me here. how is it working property? it's outdated and confusing is not it?

duhafnusa4, 2021-07-13:
I meant the physical link working and not dead

chuckwagoncomputing, 2021-07-13:
@duhafnusa4 So your opinion is that the "moved to" links in the SF repo are enough, and the repo should not be removed?

duhafnusa4, 2021-07-13:
I don't have an opinion. I just said the link worked. I also just want this ticket to be completed. Can you assist?

chuckwagoncomputing, 2021-07-14:
As far as I can tell as a relative newcomer, the status quo in this project is that open issues are not of negative value.
I think this ticket only gets closed if the SF repo gets deleted.
My opinion is that it should be deleted, but the situation is slightly better now than it was 6 months ago, as our SEO has somehow improved.

mi-hol, 2022-12-06:
@rusefillc old files on [sourceforge](https://sourceforge.net/projects/rusefi/files/) should be removed. I'd be happy to own this cleanup, assuming login is provided

[image]

mi-hol, 2022-12-10:
> @rusefillc old files on [sourceforge](https://sourceforge.net/projects/rusefi/files/) should be removed

to be clear I mean "rusefi.r46.zip" with "old files" because SF refers to this as "Download Latest Version"

or

SF feature "sync file with GitHub" feature could be used to have the same versions available on both repos.
This would provide a fail-safe copy via 2 different access paths

rusefillc, 2023-01-28:
One file removed from https://sourceforge.net/projects/rusefi/files/

closed 2023-01-28 by rusefillc