rusefi/rusefi_documentation #102: We should link to the releases from the website and not to the daily builds.

rusefillc, 2020-09-18:
@stefanst says We should link to the releases from the website and not to the daily builds.

stefanst, 2020-09-18:
Maybe some explanatory reasoning is in order:
The daily builds are experimental in character. They contain the latest and greatest ideas and, at times, they suffer from being broken to some degree. For me, as somebody who builds units for 3rd parties (i.E. customers), I need to be able to send them to a location where they can find the stable release that was used to program their specific unit. They usually won't care about the latest and greatest experimental feature. A rusEFI that actually runs their car seems more important. 
This would also make it easier to link to the proper Tunerstudio config (rusefi.ini) for their build. 
While all this can, of course, be accomplished by sending them to the link to the stable FW in the forum thread, I think the website is the more obvious choice. Otherwise questions as to "why not use the latest release from the website" will inevitably occur.   
The website should also link to previous "stable" releases. It is not always necessary for somebody who has a perfectly running vehicle to upgrade to the latest stable FW and potentially introduce new issues. Yet they may need to install Tunerstudio on a new device, or maybe they want an exact copy of their rusEFI for another vehicle etc.

OrchardPerformance, 2020-09-18:
I would tend to agree with that. I think all the dev guys know its /build_server for the daily builds and we can link it from the wiki for those who don't remember it. 
I have been shipping all units out with the latest release build and suggesting they get the release from github unless they have a specific issue a daily build fixes.

rusefillc, 2020-09-19:
We all are in agreement

@OrchardPerformance has all the powers to edit both legacy and current wiki :)

rusefillc, 2020-09-20:
github has smart linking to latest release artifacts like https://github.com/rusefi/rusefi/releases/latest/download/rusefi_bundle_mre_f4.zip

rusefillc, 2020-09-20:
https://github.com/rusefi/rusefi/wiki/Download has links mostly to latest now

stefanst, 2020-09-25:
Customers with existing boards may at times want to be able to download previous releases. We should have a collection of links to all stable releases somewhere. At this time all I can find is links to the latest release.

rusefillc, 2020-09-27:
@stefanst how about now?

https://github.com/rusefi/rusefi/wiki/Download

duhafnusa4, 2021-07-22:
@stefanst can you please comment. Its been 10 months since request.

closed 2022-12-11 by mi-hol

mi-hol, 2022-12-11:
remaining task in #266