rusefi/rusefi_documentation #105: Pressures should be clear if they are absolute or gauge

OrchardPerformance, 2020-09-28:
Some confusion arising at to the scale of the pressures we use.
Suggest we need to clarify in the documents or mark the units as kPa(a) AND kPa(g) to signify absolute and gauge.

Maybe we should also standardise to absolute?

duhafnusa4, 2021-07-13:
@mck1117 are the values displayed in TS abs?

duhafnusa4, 2021-07-22:
@mk1117 poke

mck1117, 2021-07-22:
I'm probably wrong, but I think all pressures displayed are absolute

blundar, 2021-07-22:
IMHO, there is no reason for there to be anything other than psia in the
firmware.  If humans whine too loudly, figure out a way to go PSIa to PSIg
in TunerStudio.
-D

mck1117, 2021-07-22:
dave, what's a a pound and what's an inch

blundar, 2021-07-22:
Touche.  Kpa / Mpa FTW.
-D

blundar, 2021-07-22:
IN RUSSIA, VE USE ZE METRIC SYSTEM

duhafnusa4, 2021-07-23:
OK.....so whats the consensus here? I would like if someone (cough cough papa A) would declare the choice.

kpa(a) seems the obvious choice...

mi-hol, 2022-12-11:
Was this resolved or does it need further activities?