rusefi/rusefi_documentation #107: google page index shows legacy wiki :(

rusefillc, 2020-11-18:
part of the issue is that legacy wiki is at rusefi.com domain while new wiki is at github domain so google does not see new GH wiki as part of rusEFI

[image]

duhafnusa4, 2021-07-13:
no wiki shows up anymore at all. Is that solution? 

People can click and go to rusefi.com and then go to wiki

[image]

rusefillc, 2021-07-13:
Not solution

What we need is for rusEFI documentation to show up - which is does not at the moment.

@mck1117 any fresh rumours?

rusefillc, 2021-10-07:
oh shit

we have the file for a while https://github.com/rusefi/rusefi/blob/master/misc/www/robots.txt

but it was not uploaded to actual www since upload is manual :(

anyway let me also make it multi-line

rusefillc, 2021-10-07:
We now have https://rusefi.com/robots.txt

closed 2021-10-07 by rusefillc