rusefi/rusefi_documentation #111: two feature lists how do we make sure those are in sync?

rusefillc, 2020-12-04:
Also boost and launch are now available we need lists to reflect

[image]

rusefillc, 2020-12-04:
@ksmola can we improve "in development" icon to be more obvious? maybe shady checkbox?

rusefillc, 2020-12-04:
I've promoted Launch and Boost to "working" since we, well, at least have unit tests :)

I've also moved all not supported towards the bottom of lists

Still have a problem of two unrelated lists and manual sync, also wiki does not have "in development" state which www.rusefi.com has, we need a solution probably?

duhafnusa4, 2021-07-22:
Current rusefi.com

[image]

Current GH wiki

[image]

Looks like rusefi.com needs to be updated. I just wanted to post current status of the 2 pages.

mi-hol, 2022-11-13:
@rusefillc should this be closed?

mi-hol, 2023-05-30:
source docs: 

1. for GH WIKI is "rusefi_documentation/Home.md"
2. for rusefi.com seems not available to public ( repo "rusefi/website" is empty)

@rusefillc is this understanding correct?
In case it is I'm stuck again.

mi-hol, 2023-05-30:
source docs for rusefi.com "web_backend/www/index.html"
Is it feasible to reference a md file from repo rusefi_documentation in this source doc?

mi-hol, 2023-06-02:
@rusefillc as my question "Is it feasible to reference a md file from repo rusefi_documentation in this source doc?" had no response I plan to add a "test" menu and page to www.rusefi.com.

1. Is that ok ?
or 
2. is there a better approach for trying to reference a md file from repo rusefi_documentation in www.rusefi.com?

mi-hol, 2023-06-11:
@rusefillc may I remind about my open question above?

I'll wait for another week, then close in case no answer received

rusefillc, 2023-06-11:
I am not aware of any off-the-shelf solutions for this issue.

Some sort of a toolset would need to be created to generate both html and .md from the same feature list which is pretty expensive considering the low priority. looks like manual it would be for now :(

mi-hol, 2023-06-11:
well there are several options to solve this issue mentioned in https://lowendtalk.com/discussion/67562/best-way-to-render-markdown-with-nginx and I would suggest to use [StrapDown](https://github.com/Naereen/StrapDown.js/blob/master/README.md#features-dizzy)

While I understand the low priority I believe in DRY philosophy and wont waste my time to keep 2 files in sync when a simple solution is already existing.
Would using [strapdown.js](https://github.com/Naereen/StrapDown.js/blob/master/trapdown.js) be ok for rusEFI?

rusefillc, 2023-06-11:
i am happy to review pull requests. I have nothing against .js.

chuckwagoncomputing, 2025-12-16:
It looks like there is no intent to synchronize the two lists, and in my opinion the complexity of doing so isn't worth it.