rusefi/rusefi_documentation #116: update logos

ksmola, 2021-01-01:
[rusefi_logo_vertical.tar.gz](https://github.com/rusefi/rusefi_documentation/files/5759205/rusefi_logo_vertical.tar.gz)
Here's the vertical version of the rusefi logo. 

sites that currently use the old logo and should be updated (there might be more...):
(replace with either vertical or horizontal logo, whatever is more suitable)
- github
- youtube
- facebook
- instagram

rusefillc, 2021-01-01:
need a trivial way to find files for the next 3 years. we have 'style' folder in primary repo, let's rename style to legacy style and push new style into same folder?

ksmola, 2021-01-01:
sounds like a plan, I'll PR shortly

rusefillc, 2021-01-01:
FB likes to circle does it?

[image]

rusefillc, 2021-01-01:
instagram loves circles as well

[image]

ksmola, 2021-01-01:
latest PR has wider logos

rusefillc, 2021-01-01:
[image]

[image]

rusefillc, 2021-01-01:
hehe 
[image]

ksmola, 2021-01-02:
pr'd a tiny version of the wide black-on-orange logo

rusefillc, 2021-01-02:
zero priority

shall we add "150" into the name? legacy icons had sizes in names because that was sometimes handy

[image]

duhafnusa4, 2021-07-13:
@ksmola are there any left? Can you verify complete? If so please close ticket.

duhafnusa4, 2021-07-22:
@ksmola poke.

closed 2022-12-11 by mi-hol