rusefi/rusefi_documentation #123: Broken Links

chuckwagoncomputing, 2021-02-16:
Here is the current output of brokenlinks.sh:

```
In ./Technical/Temperature-Sensing.md:
Images/analoginfo.png
Could not find
In ./Dev/Dev-Status.md:
null
Could not find
In ./Dev/Dev-Status.md:
null
Could not find
In ./Vehicles/Mini-Cooper-2003.md:
Images/2003_cooper_tach_wiring.pdf
Could not find
In ./Vehicles/Mazda-Miata-2003.md:
Images/NB2 Miata ECU Connectors Drawing.png
Could not find
In ./Spark/Multi-Spark.md:
Spark/Images/multisparkmatlab.png "Multi-spark theory"
Could not find
In ./installations/MazdaMiataNA6_Frankenso_pnp/Frankenso-MazdaMiataNA6-pnp.md:
latches_open_J_wbo_added.jpg
Could not find
In ./FAQ/FAQ-Ignition.md:
fix link
Could not find
In ./Hardware/Frankenso/Hardware-Frankenso.md:
Hardware/Frankenso/hi_side_4.jpg
Could not find
In ./Hardware/connectors/OEM-121-pin-connectors.md:
Hardware/connectors1241434_top.jpg
Could not find
In ./Hardware/Pages-Hardware.md:
microrusefi48adapter.pdf
Could not find
In ./Hardware/Completeness-Chart.md:
Hardware/Images/BareMRE.png "Bare MRE"
Could not find
In ./Hardware/Completeness-Chart.md:
Hardware/Images/PartialMRE.png "Partial MRE"
Could not find
In ./Hardware/Completeness-Chart.md:
Hardware/Images/PopulatedMRE.png "Populated MRE"
Could not find
In ./Hardware/Completeness-Chart.md:
Hardware/Images/CompleteMRE.png "Complete MRE"
Could not find
In ./_Sidebar.md:
PnP-PCB-HowTo
Could not find
In ./HOWTO/HOWTO-ask-questions.md:
Pages-FAQs
Could not find
In ./HOWTO/Pages-Software.md:
Downloads
Could not find
In ./HOWTO/Cranking.md:
Start_Stop_Button
Could not find
In ./HOWTO/Cranking.md:
Starter_Block_Relay
Could not find
In ./HOWTO/Cranking.md:
Trigger_Wheels
Could not find
In ./HOWTO/Cranking.md:
Running
     1  ./Misc/Saab_Trionic_8_Combustion Detection Module_on_Mazda_Miata_running_rusEFI
     2  ./Misc/Saab_Trionic_8_Combustion Detection Module_on_Mazda_Miata_running_rusEFI/Saab-Trionic-8-Combustion-Detection-Module-on-Mazda-Miata-running-rusEFI.md
     3  ./rusEFI-History/List-of-engines-running-rusEFI.md
     4  ./HOWTO/HOWTO-Get-Running.md

In ./Vaults/Pages-Info-Vaults.md:
Vault_BWM_Info.md
Could not find
```

duhafnusa4, 2021-07-13:
@chuckwagoncomputing can you run this again to get the latest list of dead links?

chuckwagoncomputing, 2021-07-14:
Looks like the script has some brokenness that I'll need to fix, but yeah.

chuckwagoncomputing, 2021-07-16:
Here ya go

```
In ./Technical/Temperature-Sensing.md:
Images/analoginfo.png
Could not find
In ./Dev/Dev-Status.md:
null
Could not find
In ./Dev/Dev-Status.md:
null
Could not find
In ./Vehicles/Mini-Cooper-2003.md:
Images/2003_cooper_tach_wiring.pdf
Could not find
In ./Vehicles/Mazda-Miata-2003.md:
Images/NB2 Miata ECU Connectors Drawing.png
Could not find
In ./installations/MazdaMiataNA6_Frankenso_pnp/Frankenso-MazdaMiataNA6-pnp.md:
latches_open_J_wbo_added.jpg
Could not find
In ./FAQ/FAQ-Ignition.md:
fix link
     1  ./.git/refs/remotes/origin/fix-full-links
     2  ./.git/refs/heads/fix-full-links
     3  ./.git/logs/refs/remotes/origin/fix-full-links
     4  ./.git/logs/refs/heads/fix-full-links

In ./FAQ/FAQ-Basic-Wiring-and-Connections.md:
HOWTO-electronic-throttle-body
Could not find
In ./Hardware/Hellen/Hellen72.md:
Hardware/Hellen/hellen72-d-schematic.pdf
Could not find
In ./Hardware/Frankenso/Hardware-Frankenso.md:
Hardware/Frankenso/hi_side_4.jpg
Could not find
In ./Hardware/connectors/OEM-121-pin-connectors.md:
Hardware/connectors1241434_top.jpg
Could not find
In ./Hardware/Pages-Hardware.md:
microrusefi48adapter.pdf
Could not find
In ./_Sidebar.md:
PnP-PCB-HowTo
Could not find
In ./HOWTO/HOWTO-ask-questions.md:
Pages-FAQs
Could not find
In ./HOWTO/Pages-Software.md:
Downloads
Could not find
In ./HOWTO/Cranking.md:
Start_Stop_Button
Could not find
In ./HOWTO/Cranking.md:
Starter_Block_Relay
Could not find
In ./HOWTO/Cranking.md:
Trigger_Wheels
Could not find
In ./HOWTO/Cranking.md:
Running
     1  ./Misc/Saab_Trionic_8_Combustion Detection Module_on_Mazda_Miata_running_rusEFI
     2  ./Misc/Saab_Trionic_8_Combustion Detection Module_on_Mazda_Miata_running_rusEFI/Saab-Trionic-8-Combustion-Detection-Module-on-Mazda-Miata-running-rusEFI.md
     3  ./OEM-Docs/Nissan/2011_Xterra/2011-nissan-CAN-June-2021/engine-not-running.trc
     4  ./List-of-Engines-Running-rusEFI.md
     5  ./HOWTO/HOWTO-Get-Running.md

In ./Vaults/Pages-Info-Vaults.md:
Vault_BWM_Info.md
Could not find
```

duhafnusa4, 2021-07-16:
Ill work my way through these and lets plan to run the script again in 1 week.

mi-hol, 2022-12-11:
todo: run report again
see https://github.com/rusefi/rusefi_documentation/blob/master/wiki-tools/brokenlinks.sh

mi-hol, 2022-12-11:
@chuckwagoncomputing is there a check for "file not referenced"?
This would avoid orphaned content to go unnoticed like in #261

mi-hol, 2022-12-12:
```
$ ./wiki-tools/brokenlinks.sh
In ./Dev/Dev-Status.md:
null
Could not find
In ./Dev/Dev-Status.md:
null
Could not find
In ./FAQ/FAQ-Basic-Wiring-and-Connections.md:
HOWTO-electronic-throttle-body
Could not find
In ./FAQ/FAQ-Ignition.md:
fix link
Could not find
In ./Hardware/Frankenso/Hardware-Frankenso.md:
Hardware/Frankenso/hi_side_4.jpg
Could not find
In ./Hardware/Hardware.md:
FAQ-Why-No-Self-Assembly/
     1  ./FAQ/FAQ-Why-No-Self-Assembly.md
```
Note: script was running for a long time (>30 mins) but not exiting back to prompt
ToDo: investigate further

mi-hol, 2022-12-12:
tried https://github.com/webhintio/markdown-link-validator
findings:

- debug mode for troubleshooting is extremely helpful
- it's incredible fast 
-  https://www.te.com/usa-en/ seems to block or throttle access via an automated script after a certain number of request
```
  markdown-link-validator:utils:request problem with request: connect ETIMEDOUT 2a01:4f8:221:3b0a:1000:0:19:200:443 - https://www.lammertbies.nl/comm/cable/RS-232.html +21s
  markdown-link-validator:utils:request problem with request: read ECONNRESET - https://www.te.com/usa-en/product-4-1437290-1.html +1m
```

solution:

- use exclude list for batch validation
```
markdown-link-validator . -i https?:\/\/.* --debug >.\wiki-tools\link_validation.log
```

mi-hol, 2022-12-13:
\rusefi_documentation\Dev\Dev-Status.md
Γ£û #Base-Engine:19:15
Γ£û #Fuel:20:8
Γ£û #Ignition:21:12
Γ£û #Cranking:22:12
Γ£û #Idle:23:8
Γ£û #Advanced:24:12
Γ£û #Sensors:25:11
Γ£û #Controller:26:14
Γ£û #Vaporware:27:29

rusefi_documentation\HOWTO\FSIO.md
Γ£û ##More-Examples:151:19
Γ£û ##FSIO-parameters:152:21
Γ£û ##FSIO-digital-inputs:153:25
Γ£û ##FSIO-analog-inputs:154:24
Γ£û ##Case-study-#1:-shift-light:155:32
Γ£û ##Case-study-#2:-A/C-compressor-control:156:43
Γ£û ##Case-study-#3:-digital-inputs-for-extra-devices:157:53
Γ£û ##Case-study-#4:-boost-controller:158:37
Γ£û ##Case-study-#5:-solenoid-exhaust-cam-actuator:159:50
Γ£û ##See-aslo:160:15
Γ£û ##Pin-naming-usability-issue:161:33

mi-hol, 2022-12-15:
fixed files:
Overview\TS_generated\rusEFI-project-Controller.md #277

mi-hol, 2022-12-15:
fixed all issues detected by  markdown-link-validator

\rusefi_documentation\wiki-tools\md_link_validation.log
```
rusefi_documentation> markdown-link-validator . -i https?:\/\/.* --debug >.\wiki-tools\md_link_validation.log
...
Found a total of 1363 links in directory "C:\Users\mholzer\Documents\git\rusefi_documentation":
    1363 valid
    0 invalid
Time to validate: 0.751s
```

mi-hol, 2022-12-17:
Reopen due to https://github.com/rusefi/rusefi_documentation/issues/293 causing same issue again

mi-hol, 2022-12-22:
link works on GitHub but not on wiki3
[image]

chuckwagoncomputing, 2022-12-22:
Several issues here:
Edit on Github links: MkDocs sidebar not update for flat structure
Links and images on Cranking page: HTML links

mi-hol, 2022-12-22:
https://sitechecker.pro/
[image]

mi-hol, 2022-12-22:
[image]

mi-hol, 2022-12-22:
[image]

mi-hol, 2022-12-22:
pages with 404
[image]

chuckwagoncomputing, 2022-12-23:
@mi-hol Can you run the sitechecker tool again?

mi-hol, 2022-12-23:
[image]

[image]

[image]

[image]

[image]

chuckwagoncomputing, 2022-12-23:
It seems like it didn't run again, just stored the results from last time.

mi-hol, 2022-12-23:
[image]

mi-hol, 2022-12-24:
> It seems like it didn't run again, just stored the results from last time.

actually I had "recrawled" the site and got the email below
[image]

clicking the link I got

[image]

expanding section I got:
[image]

5 day for testing of sitechecker are left for me.
Maybe you wanted to sign-up for a 7 days trail of sitechecker after my trial period expired?

chuckwagoncomputing, 2022-12-24:
It appears that it's still remembering pages that have no link to them anymore. They still exist because the workflow that uploads wiki3 overwrites without removing what already exists in order to not make the wiki go down temporarily.
@rusefillc I can change this, or you can delete the directory contents manually this time, then make a change so that the workflow runs.

rusefillc, 2022-12-24:
I was about to wipe out the folder and then I've noticed .htaccess 404 and CNAME files which I suspect I should not wipe out, now I am a bit scared about wiping out the folder blindly.

Any chance for a manual-only GHA action to remove only .md and .png? Would you rather I only remove subfolders completely and keep files in root folder? Still want me to wipe out completely? Else?

chuckwagoncomputing, 2022-12-24:
Oh yes I forgot about those. You should be able to delete everything except for .htaccess.
404 and CNAME are included in the uploaded files.

rusefillc, 2022-12-24:
Deleting 454 folders now...

rusefillc, 2022-12-24:
@chuckwagoncomputing 454 folders removed.

mi-hol, 2022-12-24:
Rebuilding site resulted in great improvement!

[image]

mi-hol, 2022-12-27:
Pro: 
* Website Score improved by 4 to 62
* Crawled pages increased +322
* Google indexed pages increased +92

Con:
* many new errors reported in added crawled pages
[image]

mi-hol, 2022-12-31:
@rusefillc mkdocs build reports below errors.
Apparently the referenced pictured files are missing since a long time.
Maybe you are in a position to provide the missing files or correct the docs?

WARNING  -  Documentation file 'Frankenso-MazdaMiataNA6-pnp.md' contains a link to 'latches_open_J_wbo_added.jpg' which is not found in the documentation files.
WARNING  -  Documentation file 'Hardware-Frankenso.md' contains a link to 'Hardware\Frankenso\hi_side_4.jpg' which is not found in the documentation files.

rusefillc, 2022-12-31:
I've made https://github.com/rusefi/rusefi_documentation/commit/a669b95b7a555b2fa97a9964aff166e2b7656d4d and https://github.com/rusefi/rusefi_documentation/commit/233bef0700f000eebbf4a009153ef02c525c3731 changes

where is Frankenso-MazdaMiataNA6-pnp.md mentioned in navigation?

mi-hol, 2023-01-01:
Mentioned in rusefi_documentation\_Sidebar.md
 

Gesendet: Samstag, 31. Dezember 2022 um 19:22 Uhr

 

I've made a669b95 and 233bef0 changes

mi-hol, 2023-01-02:
found a few incorrect image references, re-opening
search string "](/"

PR #335 (2023-01-02): fix incorrect image link "](/"
fixes: #123
closed 2023-01-02 by mi-hol