rusefi/rusefi_documentation #130: MRE pinout documentation inconsistent

SVeilleux9, 2021-02-22:
The pinout documentation for the MRE is inconsistent. 

Looking here the 12 V main relay pin is in the bottom right. https://github.com/rusefi/rusefi/wiki/Hardware-microRusEFI-wiring This pin is pin 4-M of the Molex connector where as the top right pin is pin 1-M of the Molex connector. 

The pinout documentation found here https://rusefi.com/docs/pinouts/microrusefi/ shows that the 12 V main relay is the top right pin, pin 1-M on the Molex connector.

These documents should be consistent, should pin 1 on the documentation be pin 1-M on the Molex connector or should pin 1 be the 12 V main relay pin?

SVeilleux9, 2021-02-22:
Is there a reason to not use the Molex markings? Instead of using pin 39 it would be pin 3C?

Also 2 pin numbers are missing on this site currently, Pin 23 and 24. Also Pin 27 and 28 are shifted one column to the right. https://rusefi.com/docs/pinouts/microrusefi/ This may be intentional but may be confusing to the end user.

chuckwagoncomputing, 2021-02-22:
Looks like pins 23 and 24 are missing because 27 and 28 are covering them up. My dumb self made a mistake, and my blind self somehow didn't notice.

Regarding pin numbers, it seems rusEFI has been using some standard that doesn't match the markings on the connector. The question is whether we continue with this standard, or change it to match the connector. It will be easier to fix the automated pinout to match our standard.

SVeilleux9, 2021-02-22:
What will be easiest for the end user to understand? In my opinion having pin 1 next to the pin 4 marking is asking for someone to make a mistake.

rusefillc, 2021-02-22:
not matching moles naming was a mistake. it is now too late to fix that mistake on my opinion, we will have to live with it

rusefillc, 2021-02-22:
we should pit a huge red warning on the pinout wiki document

SVeilleux9, 2021-02-22:
Not to pry and I will agree with you that it is too late, but is it really? How much documentation uses the previous naming convention, and how many units are out in the wild this would effect? 

If you take a look at this from a new users mindset, how confusing will using the Molex naming be? I can see this being confusing if someone develops on the board, or makes a PnP adapter board.

rusefillc, 2021-02-22:
@SVeilleux9 I would be happy to sell all MRE stock and transfer all repo ownership and not spend another minute on MRE

I do not see any other scenario where I can afford that fix. so far 1 confusion is 2% of user base, does not justify 8 hours to fix and unconfuse kicad names versions

chuckwagoncomputing, 2021-02-22:
If we were to correct everything to match the connector stampings, that would break people's TunerStudio configs, would it not?

rusefillc, 2021-02-22:
yes it would

rusefillc, 2021-02-22:
https://github.com/rusefi/rusefi/wiki/Hardware-microRusEFI-wiring#warning-for-historical-reasons-microrusefi-pin-numbers-do-not-match-molex-numbers-on-the-plastic

rusefillc, 2021-02-22:
[image]

duhafnusa4, 2021-07-13:
pinouts match and are good. Closing ticket.

closed 2021-07-13 by duhafnusa4