rusefi/rusefi_documentation #135: "looking for developers" message on boards

rusefillc, 2021-05-13:
We need a really short mix of https://github.com/rusefi/rusefi/wiki/Mission-Statement and https://github.com/rusefi/rusefi/blob/master/misc/Strategy_2021.md spelled out on some or many boards

@blundar can you articulate the "We are looking for software developers" in few words?

duhafnusa4, 2021-07-13:
@blundar this was asked 2 months ago. If not possible please note here so we can have someone else do it. Thanks!

duhafnusa4, 2021-07-13:
@rusefillc I recommend putting "developers needed!" in admins and others forum signature linking to strategy.

rusefillc, 2021-07-16:
@duhafnusa4 great idea, done!

[image]

closed 2021-07-16 by rusefillc