rusefi/rusefi_documentation #144: What shall we do with legacy wiki?

rusefillc, 2021-05-24:


chuckwagoncomputing, 2021-05-24:
It should at least redirect to the new wiki home page. Anything beyond that is manual labor. I'd suggest having a system in place for it, even if that's just @rusefillc editing his .htaccess to manually add a specific redirect.

rusefillc, 2021-05-24:
sounds like that's an argument for NOT using legacy rusefi.com/wiki path for "new" wiki

keep rusefi.com/wiki for legacy compatibility and have latest wiki somewhere else

and "somewhere else" could mean wiki.rusefi.com at github.io without SSH and stuff and we have to carefully weigh all pros and cons of wiki.rusefi.com at github.io

rusefillc, 2021-10-07:
@mck1117 say deny it via robots

i like it

mck1117, 2021-10-07:
```
User-agent; *

Disallow: /wiki/
```

closed 2022-11-14 by rusefillc

rusefillc, 2022-11-15:
https://rusefi.com/wiki is just a redirect HTML now https://github.com/rusefi/web_backend/tree/master/www/wiki