rusefi/rusefi_documentation #145: What's up with "Home" redirect?

rusefillc, 2021-05-24:
can we simply add an index option redirect file manually?

which gets me to the question - how do we handle page removal at the moment? do we just keep uploading new pages and page rename in git would give as a stale old page and new page under new name in the generated space?

chuckwagoncomputing, 2021-05-24:
I'm not sure what you mean by "index option redirect file". What I expect you to do, (unaware of a better solution) is to add a redirect to your .htaccess file.

As to your second question, the only real option is to manually remove them, or remove the entire wiki directory and run the workflow again. If your server had SSH access we could use rsync or something and it wouldn't be a problem.

rusefillc, 2021-05-24:
"index option redirect file" was me referencing the magic of .htaccess which I am vaguely familiar with but not extremely comfortable

to keep my small brain from exploding can you please give me full .htaccess and I am happy to drop it?

Another approach would be for me to create a new SFTP user and share password with you, or else

chuckwagoncomputing, 2021-05-24:
I use Nginx not Apache so I don't have a setup in place to test to make sure it works.
And the content will depend on what URL you choose, but it will probably be something like `Redirect 301 /wiki /wiki/Home`

SFTP won't get us anything over FTP if it's _only_ SFTP. SFTP is basically SSH, but with a shell that only accepts a limited number of commands.

Actually... we CAN delete the wiki directory with FTP! I don't know why I didn't think of that. The wiki would just be kinda down while the new files are being uploaded. SSH with rsync wouldn't have that problem.

rusefillc, 2021-05-25:
apparently there was 

[image]

not 100% sure if in proper folder I am confused since other tools are showing my brand new

[image]

chuckwagoncomputing, 2022-01-22:
This is fixed now!

closed 2022-01-22 by chuckwagoncomputing