rusefi/rusefi_documentation #148: New Wiki image path drama

chuckwagoncomputing, 2021-05-25:
The new wiki doesn't handle paths quite the same as the Github Wiki.
When a mardown file is in a sub-directory, e.g. Hardware, and has an image in it that is also in a subdirectory, Github requires the path like so:
"Hardware/foo.png"
MkDocs parses this relative to the directory of the markdown file: "Hardware/Hardware/foo.png"
This is no good. MkDocs does have an option for absolute paths, e.g.
"/Hardware/foo.png"
but this doesn't work on Github.
So if/when we switch, image links will need to be rewritten. This could be automated.

Blocks #138

rusefillc, 2021-10-07:
Closing FOR NOW since the miracle of #138 has happened?

chuckwagoncomputing, 2021-10-19:
It does look like this is changed/fixed. Closing.