rusefi/rusefi_documentation #151: Case Study: we need a video showing 20 cranking/starting/engine off cycles

rusefillc, 2021-07-07:
Let's test cranking/starting reliability

rusefillc, 2021-07-07:
https://github.com/rusefi/rusefi/wiki/Case-Studies#consecutive-engine-cranking--engine-stop

BeerMoneyMotorsports, 2021-07-07:
Commenting to keep track of issue. Logs with each video?

rusefillc, 2021-07-07:
For this one the most important thing is to show 100% success rate on the video. Could be one long laptop log just fine

BeerMoneyMotorsports, 2021-07-07:
Cold or hot starts?

rusefillc, 2021-07-07:
both eventually but warm to begin with?

stefanst, 2021-07-07:
Do we need the proof for ourselves? I don't know of any customer actually doubting that if rusEFI can start the engine once it can do it again. And again. And again....

rusefillc, 2021-07-07:
I totally doubt myself for sure.

But yes, also proof for future customers and overall as part of documentation.

duhafnusa4, 2021-07-22:
rusEFI running conditions we want to document with video. 15 variations in total. This is daunting but very good data. Datalogs and tune corresponding to each condition is critical.

1) Engine Cold
    a) Starting (how long cranking till engine running)
    b) Idle (is it smooth or jumpy?, does engine stall when AC or PS is being used?)
    c)  In town/traffic driving - small throttle inputs (smooth idle, good power)
    d) Highway driving - medium throttle inputs ( maintains speed, smooth running, good lambda)
    e) Enthusiastic driving - heavy throttle inputs ( engine has good power, good lambda, off throttle decel smooth)

2) Engine Normal operating temp
    a) Starting (how long cranking till engine running)
    b) Idle (is it smooth or jumpy?, does engine stall when AC or PS is being used?)
    c)  In town/traffic driving - small throttle inputs (smooth idle, good power)
    d) Highway driving - medium throttle inputs ( maintains speed, smooth running, good lambda)
    e) Enthusiastic driving - heavy throttle inputs ( engine has good power, good lambda, off throttle decel smooth)

3) Engine Hot
    a) Starting (how long cranking till engine running)
    b) Idle (is it smooth or jumpy?, does engine stall when AC or PS is being used?)
    c)  In town/traffic driving - small throttle inputs (smooth idle, good power)
    d) Highway driving - medium throttle inputs ( maintains speed, smooth running, good lambda)
    e) Enthusiastic driving - heavy throttle inputs ( engine has good power, good lambda, off throttle decel smooth)