rusefi/rusefi_documentation #154: Standard for Icons Used

duhafnusa4, 2021-07-13:
We need to standardize Icons used for linking to other pages and whatnot. 

1) Can I just google and use or will that infringe on creator?
2) Should I create custom Icon pack? - this would be annoying
3) Does anyone have icons we can use?

rusefillc, 2021-07-13:
@ksmola is the boss

duhafnusa4, 2021-07-14:
@ksmola can you assist?

mi-hol, 2022-12-11:
@rusefillc did #116 solve this too or is there an open action? in case yes, what exactly?

rusefillc, 2022-12-11:
my understanding this is about visual improvements for pages like https://github.com/rusefi/rusefi/wiki/AlphaX-4chan and https://github.com/rusefi/rusefi/wiki/Hellen-121-Nissan which currently have too many text links

Still actual, no progress was made :(

rusefillc, 2022-12-11:
@ElDominio do you see a value? Do you have any web people to provide help with how the pages above look like?

rusefillc, 2023-01-28:
We have no resources to improve

closed 2023-01-28 by rusefillc