rusefi/rusefi_documentation #156: upload tune WAT?

rusefillc, 2021-07-17:
https://github.com/rusefi/rusefi/wiki/HOWTO_upload_tune

https://github.com/rusefi/rusefi/wiki/HOWTO-upload-tune

@duhafnusa4 where did you find https://github.com/rusefi/rusefi/wiki/HOWTO_upload_tune bad link which needs to be fixed?

@chuckwagoncomputing that's still us fixing the global page rename :(

chuckwagoncomputing, 2021-07-18:
deleted HOWTO-upload-tune

rusefillc, 2021-07-18:
@duhafnusa4 what has started all this? where is the broken link?

duhafnusa4, 2021-07-22:
Ive spent a lot of time trying to recreate the issue that stemmed from a hellen page but I cannot. For now I will close this ticket, and if I find any dead links I will make a new ticket, instead of making a new page :)

closed 2021-07-22 by duhafnusa4