rusefi/rusefi_documentation #163: Decide on what DFU tracks we support and why

rusefillc, 2021-09-26:
rusEFI console still works via legacy DFU driver.

We do not update rusEFI console to "new" driver only due to dev resources shortage.

Are there any reasons to use STM32 program or shall we stop to confuse users?

rusefillc, 2021-09-26:
At the moment we clearly document requirement to REMOVE new driver

https://github.com/rusefi/rusefi/wiki/HOWTO-DFU#q-i-have-stm32-bootloader-in-device-manager-and-rusefi-console-does-not-update-firmware-whats-wrong

rusefillc, 2021-09-26:
Bill brings the question of Windows VS not-Windows

BeerMoneyMotorsports, 2021-09-26:
I only use stmcube for initial wideband firmware install. End users should not need to use it. They should only use auto dfu and then manual dfu in case of emergency.

closed 2021-11-16 by rusefillc