rusefi/rusefi_documentation #164: FTP is unreliable

chuckwagoncomputing, 2021-09-27:
We keep getting errors like this, which leaves things broken.

```
connect failed: Connection refused.
connect failed: Connection refused.
connect failed: Connection refused.
Falling back to PORT instead of PASV mode.
pinouts/proteus/white35.jpg:    Lost data connection to remote host after 32768 bytes had been sent: Broken pipe.
```

blundar, 2021-09-28:
that could be firewall related.

rusefillc, 2021-09-28:
We now have https://rusefi.com/docs/pinouts-backup/ one time manual backup

in my opinion we need to push pinout pages into git. those are kind of same thing-ish as .ini files?

also it would be a dream to have a .pdf generated automatically

chuckwagoncomputing, 2021-10-14:
OK, so using the primary account would mean putting the authentication details or a private key in GH Secrets. Preferably a private key used only for this.

chuckwagoncomputing, 2021-10-19:
I'm not sure how you've had FTP set up, but typically FTP is scoped to a specific directory, and the rest of the file system is inaccessible. With SSH, if you don't `cd` somewhere, files will by go in the home directory. Have you confirmed that the files were indeed updated on the server?

rusefillc, 2021-10-19:
I cannot explain it but yes looks like files are updating

[image]

rusefillc, 2021-10-19:
https://rusefi.com/build_server/autoupdate/

rusefillc, 2021-10-19:
and https://rusefi.com/build_server/

rusefillc, 2021-10-21:
broken again :( https://rusefi.com/docs/pinouts/proteus/

rusefillc, 2021-11-01:
Switching from FTP to SFTP has resolved this issue

closed 2021-11-01 by rusefillc