rusefi/rusefi_documentation #172: mkdocs warnings

rusefillc, 2021-10-14:
how legit are those?

```
debian@debian:~/rusefi_documentation$ bin/mkdocs-run.sh
INFO     -  Cleaning site directory
INFO     -  Building documentation to directory: /home/debian/rusefi_documentation/mkdocs/site
WARNING  -  Documentation file 'Hardware/Frankenso/Hardware-Frankenso.md' contains a link to 'Hardware/Frankenso/hi_side_4.jpg' which is not found in the documentation files.
WARNING  -  Documentation file 'Hardware/connectors/OEM-121-pin-connectors.md' contains a link to 'Hardware/connectors1241434_top.jpg' which is not found in the documentation files.
WARNING  -  Documentation file 'Hardware/pnp_microRusEFI_48na/Miata-NA-48-pin-board-for-MRE.md' contains a link to 'Hardware/pnp_microRusEFI_48na/microrusefi48adapter_0_2.pdf' which is not found in the documentation files.
WARNING  -  Documentation file 'Hardware/pnp_microRusEFI_48na/Miata-NA-48-pin-board-for-MRE.md' contains a link to 'Hardware/pnp_microRusEFI_48na/microrusefi48adapter_0.1.pdf' which is not found in the documentation files.
WARNING  -  Documentation file 'Misc/Saab_Trionic_8_Combustion Detection Module_on_Mazda_Miata_running_rusEFI/Saab-Trionic-8-Combustion-Detection-Module-on-Mazda-Miata-running-rusEFI.md' contains a link to 'Misc/Saab_Trionic_8_Combustion
            Detection Module_on_Mazda_Miata_running_rusEFI/Misc/Saab_Trionic_8_Combustion Detection Module_on_Mazda_Miata_running_rusEFI/saab_cdm_tldr_knocking.png' which is not found in the documentation files.
WARNING  -  Documentation file 'Misc/Saab_Trionic_8_Combustion Detection Module_on_Mazda_Miata_running_rusEFI/Saab-Trionic-8-Combustion-Detection-Module-on-Mazda-Miata-running-rusEFI.md' contains a link to 'Misc/Saab_Trionic_8_Combustion
            Detection Module_on_Mazda_Miata_running_rusEFI/Misc/Saab_Trionic_8_Combustion Detection Module_on_Mazda_Miata_running_rusEFI/saab_cdm_knock_signal.png' which is not found in the documentation files.
WARNING  -  Documentation file 'Misc/Saab_Trionic_8_Combustion Detection Module_on_Mazda_Miata_running_rusEFI/Saab-Trionic-8-Combustion-Detection-Module-on-Mazda-Miata-running-rusEFI.md' contains a link to 'Misc/Saab_Trionic_8_Combustion
            Detection Module_on_Mazda_Miata_running_rusEFI/Misc/Saab_Trionic_8_Combustion Detection Module_on_Mazda_Miata_running_rusEFI/saab_cdm_tldr_not_knocking.png' which is not found in the documentation files.
WARNING  -  Documentation file 'Misc/Saab_Trionic_8_Combustion Detection Module_on_Mazda_Miata_running_rusEFI/Saab-Trionic-8-Combustion-Detection-Module-on-Mazda-Miata-running-rusEFI.md' contains a link to 'Misc/Saab_Trionic_8_Combustion
            Detection Module_on_Mazda_Miata_running_rusEFI/Misc/Saab_Trionic_8_Combustion Detection Module_on_Mazda_Miata_running_rusEFI/saab_cdm_combustion_signal.png' which is not found in the documentation files.
WARNING  -  Documentation file 'Misc/Saab_Trionic_8_Combustion Detection Module_on_Mazda_Miata_running_rusEFI/Saab-Trionic-8-Combustion-Detection-Module-on-Mazda-Miata-running-rusEFI.md' contains a link to 'Misc/Saab_Trionic_8_Combustion
            Detection Module_on_Mazda_Miata_running_rusEFI/Misc/Saab_Trionic_8_Combustion Detection Module_on_Mazda_Miata_running_rusEFI/saab_2005_parts.jpg' which is not found in the documentation files.
WARNING  -  Documentation file 'Misc/Saab_Trionic_8_Combustion Detection Module_on_Mazda_Miata_running_rusEFI/Saab-Trionic-8-Combustion-Detection-Module-on-Mazda-Miata-running-rusEFI.md' contains a link to 'Misc/Saab_Trionic_8_Combustion
            Detection Module_on_Mazda_Miata_running_rusEFI/Misc/Saab_Trionic_8_Combustion Detection Module_on_Mazda_Miata_running_rusEFI/saab_cdm_hardware_details.jpg' which is not found in the documentation files.
WARNING  -  Documentation file 'Misc/Saab_Trionic_8_Combustion Detection Module_on_Mazda_Miata_running_rusEFI/Saab-Trionic-8-Combustion-Detection-Module-on-Mazda-Miata-running-rusEFI.md' contains a link to 'Misc/Saab_Trionic_8_Combustion
            Detection Module_on_Mazda_Miata_running_rusEFI/Misc/Saab_Trionic_8_Combustion Detection Module_on_Mazda_Miata_running_rusEFI/saab_cdm_frankenso_assembled.jpg' which is not found in the documentation files.
WARNING  -  Documentation file 'Misc/Saab_Trionic_8_Combustion Detection Module_on_Mazda_Miata_running_rusEFI/Saab-Trionic-8-Combustion-Detection-Module-on-Mazda-Miata-running-rusEFI.md' contains a link to 'Misc/Saab_Trionic_8_Combustion
            Detection Module_on_Mazda_Miata_running_rusEFI/Misc/Saab_Trionic_8_Combustion Detection Module_on_Mazda_Miata_running_rusEFI/saab_cdm_engine_bay.jpg' which is not found in the documentation files.
WARNING  -  Documentation file 'Misc/Saab_Trionic_8_Combustion Detection Module_on_Mazda_Miata_running_rusEFI/Saab-Trionic-8-Combustion-Detection-Module-on-Mazda-Miata-running-rusEFI.md' contains a link to 'Misc/Saab_Trionic_8_Combustion
            Detection Module_on_Mazda_Miata_running_rusEFI/Misc/Saab_Trionic_8_Combustion Detection Module_on_Mazda_Miata_running_rusEFI/saab_cdm_knock_ignition_table.png' which is not found in the documentation files.
WARNING  -  Documentation file 'Misc/Saab_Trionic_8_Combustion Detection Module_on_Mazda_Miata_running_rusEFI/Saab-Trionic-8-Combustion-Detection-Module-on-Mazda-Miata-running-rusEFI.md' contains a link to 'Misc/Saab_Trionic_8_Combustion
            Detection Module_on_Mazda_Miata_running_rusEFI/Misc/Saab_Trionic_8_Combustion Detection Module_on_Mazda_Miata_running_rusEFI/saab_cdm_2018_01_05_miata_rigol.zip' which is not found in the documentation files.
WARNING  -  Documentation file 'Misc/Saab_Trionic_8_Combustion Detection Module_on_Mazda_Miata_running_rusEFI/Saab-Trionic-8-Combustion-Detection-Module-on-Mazda-Miata-running-rusEFI.md' contains a link to 'Misc/Saab_Trionic_8_Combustion
            Detection Module_on_Mazda_Miata_running_rusEFI/Misc/Saab_Trionic_8_Combustion Detection Module_on_Mazda_Miata_running_rusEFI/H6T60271_and_H6T61171.jpg' which is not found in the documentation files.
WARNING  -  Documentation file 'Misc/Saab_Trionic_8_Combustion Detection Module_on_Mazda_Miata_running_rusEFI/Saab-Trionic-8-Combustion-Detection-Module-on-Mazda-Miata-running-rusEFI.md' contains a link to 'Misc/Saab_Trionic_8_Combustion
            Detection Module_on_Mazda_Miata_running_rusEFI/Misc/Saab_Trionic_8_Combustion Detection Module_on_Mazda_Miata_running_rusEFI/Ionization_Detection_Module_55352173.png' which is not found in the documentation files.
WARNING  -  Documentation file 'Technical/Temperature-Sensing.md' contains a link to 'Images/analoginfo.png' which is not found in the documentation files.
WARNING  -  Documentation file 'Vaults/Pages-Info-Vaults.md' contains a link to 'Vault_BWM_Info.md' which is not found in the documentation files.
WARNING  -  Documentation file 'Vehicles/Mazda-Miata-2003.md' contains a link to 'Vehicles/Images/NB2 Miata ECU Connectors Drawing.png' which is not found in the documentation files.
WARNING  -  Documentation file 'installations/MazdaMiataNA6_Frankenso_pnp/Frankenso-MazdaMiataNA6-pnp.md' contains a link to 'latches_open_J_wbo_added.jpg' which is not found in the documentation files.
INFO     -  Documentation built in 13.98 seconds
debian@debian:~/rusefi_documentation$

```

rusefillc, 2021-10-14:
Some are legit - some were just improved

chuckwagoncomputing, 2023-01-07:
Only remaining are pages without links to them, which is just the result of the approach of throwing info into the wiki for archival/memory purposes.

closed 2023-01-07 by chuckwagoncomputing