rusefi/rusefi_documentation #176: Split firmware from hardware

rusefillc, 2021-10-30:
At the moment https://github.com/rusefi/rusefi has amazing firmware and pretty legacy hardware

Theoretically that could be sending a wrong signal.

it also contributes to cloned copy 2Gb size which could be a non-issue

rusefillc, 2021-10-30:
Kind of related to https://github.com/rusefi/rusefi/issues/3433

OrchardPerformance, 2021-10-30:
Split hardware and firmware repository into one specific for each? 
If so, 100% agree with that idea

rusefillc, 2021-10-30:
@OrchardPerformance at this point I am not sure how we would implement the split. there is a lot of GIT magic details

rusefillc, 2021-10-30:
option 1:  new https://github.com/rusefi/rusefi repo, bad option because history loss

* rename https://github.com/rusefi/rusefi to https://github.com/rusefi/rusefi-hardware
* copy master of https://github.com/rusefi/rusefi without hardware into 
* delete firmware from https://github.com/rusefi/rusefi-hardware

option 2: move hardware content to new repo
* copy-paste https://github.com/rusefi/rusefi/hardware into new repo https://github.com/rusefi/rusefi-hardware
* delete hardware from master
* I guess having hardware history in https://github.com/rusefi/rusefi is not horrible?

OrchardPerformance, 2021-10-31:
I like option 2, but I am an idiot. 

I think the history showing there used to be hardware but now it has moved and there is newer hardware is fine.

rusefillc, 2021-11-01:
a bunch of hardware (but not all) moved to https://github.com/rusefi/rusefi-hardware/tree/main/classic-designs

closed 2021-11-01 by rusefillc