rusefi/rusefi_documentation #177: TS Online Tuning links to empty page

duhafnusa4, 2021-11-02:
on https://github.com/rusefi/rusefi/wiki/ the linked "TunerStudio online tuning" takes you to 

https://github.com/rusefi/rusefi/wiki/wiki

which shows "create a new page"

duhafnusa4, 2021-11-02:
launch control does the same thing

duhafnusa4, 2021-11-02:
Acceleration enrichment - wall wetting/fuel trip/tau factor does the same thing

duhafnusa4, 2021-11-02:
Digital Dash does the same thing

duhafnusa4, 2021-11-02:
https://github.com/rusefi/rusefi/wiki/ is big broken @rusefillc

BeerMoneyMotorsports, 2021-11-02:
"tuner studio online tuning" takes me to https://github.com/rusefi/rusefi/wiki/HOWTO-create-tunerstudio-project which needs to be updated and improved.

rusefillc, 2021-11-02:
This seems related to https://github.com/rusefi/rusefi_documentation/pull/147 and https://github.com/rusefi/rusefi_documentation/issues/146

@chuckwagoncomputing do you understand root cause of drama? as is I do not understand why 

```
[TunerStudio online tuning](HOWTO-create-tunerstudio-project)|✓
```
becomes

https://github.com/rusefi/rusefi/wiki/wiki/HOWTO-create-tunerstudio-project

(with double "wiki")

chuckwagoncomputing, 2021-11-02:
The only ones that aren't broken on the home page are the ones that are hard-linked, which is bad and will need to be fixed when/if we move to wiki3.
As to the cause I have no idea, but it sure looks like a Github bug.

rusefillc, 2021-11-02:
@chuckwagoncomputing any idea where do we report/inquire about that one?

chuckwagoncomputing, 2021-11-02:
I edited the page but didn't actually change anything and it seems to have re-generated properly.

chuckwagoncomputing, 2021-11-02:
Wait I take that back. After editing it took me to https://github.com/rusefi/rusefi/wiki , which works. https://github.com/rusefi/rusefi/wiki/ does not work. (note the trailing slash)

rusefillc, 2021-11-02:
We need to ket GH know either way?

chuckwagoncomputing, 2021-11-02:
Not really, that's just how links work. The real problem is that we have a link with a trailing slash somewhere.

chuckwagoncomputing, 2021-11-02:
No I take that back. The links don't need the preceding "wiki/" to work.

chuckwagoncomputing, 2021-11-02:
https://github.com/github/feedback/discussions/7157

mi-hol, 2022-12-11:
@chuckwagoncomputing is this still an open issue?

chuckwagoncomputing, 2022-12-11:
Seems to be working properly.

closed 2022-12-11 by chuckwagoncomputing