rusefi/rusefi_documentation #178: Hellen wiki standard format

BeerMoneyMotorsports, 2021-11-10:
Hellen wikis should have a standard format. Hellen 64 suitable?  

https://github.com/rusefi/rusefi/wiki/Hellen64-Miata-NA6-94

rusefillc, 2021-11-10:
https://github.com/rusefi/rusefi/wiki/Hellen64-Miata-NA6-94 is not as compact as https://github.com/rusefi/rusefi/wiki/Hellen-128-Mercedes

two lines for download links instead of one
does not have versions for revisions

in my opinion in some (definitely not all) regards https://github.com/rusefi/rusefi/wiki/Hellen-128-Mercedes is better

rusefillc, 2021-11-10:
just realized that https://github.com/rusefi/rusefi/wiki/Hellen64-Miata-NA6-94 has cool idea of schematics next to pictures I like it!

maybe ibom links should follow suit?