rusefi/rusefi_documentation #186: syntax differences or else?

rusefillc, 2021-11-16:
why are download links so broken at https://wiki.rusefi.com/Hardware/Proteus/Proteus-Manual/

[image]

[image]

chuckwagoncomputing, 2021-11-16:
Wierd syntax fixed:
https://github.com/rusefi/rusefi_documentation/commit/1b90d48c4886c326f246e5ea3e70dcb2c6c92a72

rusefillc, 2021-11-16:
I've used [[ x ]] on many Hellen landing pages see https://wiki.rusefi.com/Hellen-154-Hyundai/ and other pages linked from https://wiki.rusefi.com/Hellen-One-Platform/ I will need to revist

closed 2021-11-17 by rusefillc