rusefi/rusefi_documentation #188: auto-detect URLs

rusefillc, 2021-11-17:
https://github.com/rusefi/rusefi/wiki/Release-Process renders links as links

https://wiki.rusefi.com/Release-Process/ does not :(

is this something we can improve globally or would I have to use full syntax?

rusefillc, 2021-11-17:
is this something https://github.com/orbikm/mkdocs-ezlinks-plugin is supposed to be helping with?

chuckwagoncomputing, 2022-01-22:
No I don't think so. I suppose we could manually change them all to the correct URL format, as that would be backwards compatible with Github wiki. Maybe I'll write a script to do that.

chuckwagoncomputing, 2022-01-25:
Massive effort ongoing: https://github.com/chuckwagoncomputing/rusefi_documentation/tree/linkify

chuckwagoncomputing, 2022-02-06:
I think all of these are fixed. The title of this issue is "auto-detect URLs", which as far as I can tell isn't possible currently. I suppose it would be possible to write a MkDocs plugin that would do that, but I think it's better to manually make URLs links, and give them a proper title.

rusefillc, 2022-11-08:
Status update:

@chuckwagoncomputing has explicitly converted all URLs into full blown URL syntax like https://github.com/chuckwagoncomputing/rusefi_documentation/commit/a6c27028f406ee32294e3fe43f1cf32d8edac642

it sounds like the proposal is to just always to use explicit syntax which is nasty but gets us forward

rusefillc, 2022-11-14:
Reopening since I think it's far from done

Problem:
I am a lazy human and I've created new content on wiki2-technical https://github.com/rusefi/rusefi_documentation/blob/master/Gen-4-LS.md note how links are nicely rendered as URL

and now same new content on wiki3 https://wiki.rusefi.com/Gen-4-LS/

note how links are not URLs

I really do not like working hard. We need automation for edits on wiki2-technical or wiki2-human to be rendered as URLs on wiki3

chuckwagoncomputing, 2022-11-14:
1. "Automatic Links" are not part of the Markdown standard, they're an additional thing Github has done.
2. Bare URL links are ugly and uncivilized
3. Don't be lazy

rusefillc, 2022-11-14:
> 3. Don't be lazy

I hope to stay in human form for 30ish more years, there is no way I can take manual self-control of URL approach

if we are unable to automate the translation we have to at least automated detection: there should be at least an email or a URL where all "bad" links are mentioned so that manual human replacement could be done based on the auto-generated problem report and not just by manual review

mi-hol, 2022-11-15:
> 1. "Automatic Links" are not part of the Markdown standard, they're an additional thing Github has done.

GitHub's user experience for editors is therefore much better than what a  later wiki user gets to see.
From my view its a major gap that needs to be closed in order to get better rusEFI community involvement in  documentation

mi-hol, 2022-11-15:
I found a very similar issue with a solution in https://github.com/community/community/discussions/22154#discussioncomment-3235674
Would mkdoc perhaps have an option for GitHub Flavoured Markdown?

rusefillc, 2022-11-15:
Also question if we shall move to https://kramdown.gettalong.org/ engine ? :)

mi-hol, 2022-11-15:
> move to https://kramdown.gettalong.org/ engine

would have been my second question, in case first is answered with a "no"

chuckwagoncomputing, 2022-11-15:
After some more research, it seems that Mkdocs can pass plugins to Python-markdown, so maybe a combination of https://github.com/selcuk/markdown-urlify and https://github.com/Python-Markdown/github-links could be made to work.

And Kramdown is just a markdown engine, it only does fraction of what Mkdocs does.

chuckwagoncomputing, 2022-11-22:
I'm a noob as well.
The mkdocs config is [here](https://github.com/rusefi/rusefi_documentation/blob/master/mkdocs/mkdocs.yaml).
The documentation for the markdown_extension option is [here](https://www.mkdocs.org/user-guide/configuration/#markdown_extensions).

PR #336 (2023-01-03): Automatic URL linking - Wiki3
I found a markdown extension that does the magic!
Fixes #188
closed 2023-01-03 by rusefillc