rusefi/rusefi_documentation #189: guide robots away from https://rusefi.com/docs/wiki/Home/

rusefillc, 2021-11-28:
robots should enjoy https://wiki.rusefi.com/ not https://rusefi.com/docs/wiki/Home/

https://www.google.com/search?q=You+have+found+the+rusEFI+wiki+main+page

[image]

rusefillc, 2021-11-30:
https://rusefi.com/robots.txt updated with new disallow

open question what's best way to mention https://wiki.rusefi.com/

@mck1117 @chuckwagoncomputing any ideas?

rusefillc, 2021-12-09:
has been 9 days since robots.txt update

https://www.google.com/search?q=You+have+found+the+rusEFI+wiki+main+page still goes not where I want it to go, and does not go where I want it to go

@chuckwagoncomputing @mck1117 any ideas?

chuckwagoncomputing, 2022-01-13:
Why not redirect https://rusefi.com/docs/wiki/Home/ to https://wiki.rusefi.com/Home/?

rusefillc, 2022-01-13:
@chuckwagoncomputing I love this proposal but I have no idea how to implement. Here's the current pretty confusing setup:

1) 
subdomain points into sub-folder of primary domain
[image]

2) in order for https://wiki.rusefi.com/ to work without requiring "home" there is already a RewriteRule which is pretty off - it's place in root folder of MAIN domain  (URL of GH copy is https://github.com/rusefi/rusefi/blob/master/misc/www/.htaccess) and it does

```
RewriteRule docs/wiki$  /Home [R]
RewriteRule docs/wiki/$ /Home [R]
```

within those two constrains I have no idea how to implement your idea which I definitely like. Actually correction - it should be " https://rusefi.com/docs/wiki/Home/ to https://wiki.rusefi.com" without /Home

chuckwagoncomputing, 2022-01-14:
1. I'm not sure how this works internally but I think that's fine.
2. I'm not sure this is doing what you think it's doing. https://rusefi.com/docs/wiki/ redirects to https://rusefi.com/Home, which is a 404. The 404 page is set to the old wiki 404 page.

I suggest replacing those RewriteRules with a redirect to wiki.rusefi.com for starters, and see if that behaves properly.
Do you have access to the .htaccess file for wiki.rusefi.com?

rusefillc, 2022-01-15:
Sounds like it's time to make amazing progress here!

step 1 - DONE - https://github.com/rusefi/rusefi/issues/3790
step 2 - to answer your question main .htaccess is https://github.com/rusefi/web_backend/blob/master/www/.htaccess and by extension wiki.rusefi.com file is https://github.com/rusefi/web_backend/blob/master/www/docs/wiki/.htaccess
fun fact 1 - sync is still manual. Ticket created to automate upload from GHA to 1and1 see https://github.com/rusefi/web_backend/issues/146
fun fact 2 that means we would have two sources both uploading into same 1and1 www/docs/wiki folder, the new static content upload job and existing markup generator
fun fact 3 I need to make a backup of www and place it to the secret location before we merge https://github.com/rusefi/web_backend/issues/146
fun fact 4 once fun facts 1 and 3 are ready i would love to simply grant @chuckwagoncomputing write access to https://github.com/rusefi/web_backend/

rusefillc, 2022-01-21:
fun fact 3 status update: All www files were backed up

between https://github.com/rusefi/web_backend/issues/146 and @chuckwagoncomputing having write access David do you want to play with .htaccess files?

chuckwagoncomputing, 2022-01-22:
Done!

closed 2022-01-22 by chuckwagoncomputing