rusefi/rusefi_documentation #192: Auto generate single document of TS names and tip/hint text.

OrchardPerformance, 2021-12-09:
Had a thought to improve our documentation with minimum effort.

Can we use the same process that generates the rusefi.ini to create a .txt document that contains just the names of all the options in TS and its associated tip/hint text.

Thinking this is an automatically updated master list of all fields and their function for quick reference.

rusefillc, 2021-12-09:
@mck1117 what's your take on this overall?

Did you say you have a new approach to code generation in the works?

mck1117, 2021-12-09:
you mean like the `SettingContextHelp` section?

[image]

mi-hol, 2022-12-11:
@OrchardPerformance how shall we proceed with your suggestion?

OrchardPerformance, 2022-12-11:
It was an idea no more, the answer is we probably don't.