rusefi/rusefi_documentation #194: Joint RE, BMM, H1 ecu logo

BeerMoneyMotorsports, 2022-01-08:
I tried and failed to make a good looking combined logo.  @duhafnusa4 kindly offered to step in an help.  
[image]

duhafnusa4, 2022-01-08:
I'll work on it this week and have designs ready to review Wednesday

BeerMoneyMotorsports, 2022-01-14:
@duhafnusa4 Wednesday was two days ago........tsk tsk

duhafnusa4, 2022-01-14:
Yeah because marks designer said he was doing it? If not I'll get it done today

duhafnusa4, 2022-01-14:
What do you think about this? This is a draft obviously, but is this the direction you want to go?

[image]

BeerMoneyMotorsports, 2022-01-14:
That has some potential i think.  Im not sure if @rusefillc or @andreika-git want to give any input?  

I want to make sure people BMM with rusEFI.  I get a lot of "im interested in the bmm ecu" and id like it to be "im interested in the RE ecu you distribute".

rusefillc, 2022-01-14:
I am very flexible, more or less anything works for me.

My personal take - are we sure we should be piling up logos on top of each other? What's the problem statement?
Would small "runs rusEFI firmware" on the unit and Hellen logo on the PCB solve the problem without logo on top of logo?

BeerMoneyMotorsports, 2022-01-17:
Lets drop the hellen logo because it is on the PCB.  

For the rusEFI, can we say Powered by rusEFI maybe?  maybe off the the left of the mug?

BeerMoneyMotorsports, 2022-02-01:
@duhafnusa4 maybe something like this?  In RE font?
[image]
[image]

duhafnusa4, 2022-02-01:
something like this?

[image]

duhafnusa4, 2022-02-01:
[image]

rusefillc, 2022-02-03:
too much is going on the log in my opinion. i would just have "powered by rusefi" with just black with same font just smaller type

beer money orange with border is readable with my eyes

"powered" orange with black shade is annoying my eyes specifically your eyes might vary

duhafnusa4, 2022-02-03:
So what do you think, just black? What Font did you use for your stuff @BeerMoneyMotorsports ?

BeerMoneyMotorsports, 2022-02-03:
Invert the powered colors. Unfortunately i do not know the font.

BeerMoneyMotorsports, 2022-02-28:
@duhafnusa4 Lets try the Powered line with current colors inverted and also just black.  Font can remain the same as i do not know what font was originally used.

duhafnusa4, 2022-02-28:
> @duhafnusa4 Lets try the Powered line with current colors inverted and also just black. Font can remain the same as i do not know what font was originally used.

OK sounds good.

duhafnusa4, 2022-03-27:
[image]

duhafnusa4, 2022-03-27:
[image]

closed 2022-10-03 by BeerMoneyMotorsports