rusefi/rusefi_documentation #195: Hellen Mercedes landing page is very sad

rusefillc, 2022-01-15:
We have boards with https://rusefi.com/s/hellen128merc short URL and QR code

At the moment https://rusefi.com/s/hellen128merc is a redirect to github wiki and that github wiki page is horrible in my opinion

open question if Hellen Mercedes landing page should be github wiki or any wiki for that matter it's not like https://wiki.rusefi.com/Hellen-128-Mercedes/ is much better. Well, https://wiki.rusefi.com/Hellen-128-Mercedes/ is definitely at least somewhat better than https://github.com/rusefi/rusefi/wiki/Hellen-128-Mercedes

At the moment we have too many text links on either page. At the moment overall order of links is questionable.

I am not sure what exactly is a problem but overall it looks like current Hellen Mercedes landing page is a disaster

rusefillc, 2022-01-15:
@ksmola @shadowm60 opinions?

@BeerMoneyMotorsports has some experience with hellen landing pages

duhafnusa4, 2022-06-27:
What do you want to see happen @rusefillc ? I will work on making it nicer.

shadowm60, 2022-06-27:
I did some updates, but i have no idea how to make it nicer :( info is there, maybe hard to read it all.

mi-hol, 2022-12-11:
https://github.com/rusefi/rusefi_documentation/issues/178 and #216 seem related
@duhafnusa4 would that answer "What do you want to see happen @rusefillc ? I will work on making it nicer." at least partially?
With a first stab done, a review could be used to establish a "standard" format

mi-hol, 2023-01-13:
in text "On Rev. C hardware E1, E2, E3 and E4 can all be used as solenoid control." is mentioned but not available in "interactive pinout"
[image]

mi-hol, 2023-01-31:
description in shop is very thin too.
Should a link to wiki be added?
[image]

mi-hol, 2023-03-09:
created https://github.com/rusefi/rusefi_documentation/pull/436
Without feedback no more can be done :(