rusefi/rusefi_documentation #197: wiki clean-up for Merc content

rusefillc, 2022-01-20:
We have a bunch of random Merc schematics very poorly structured on the wiki

At the moment we mostly care about stuff related to 128 hellen - shall structure pages better?

shadowm60, 2022-01-20:
I have started this locally, progress is slow but i am on it.

abelom, 2022-01-20:
@shadowm60 what is the URL for _your_ _exact_ schematics on rusEFI wiki? :)

rusefillc, 2022-01-21:
@shadowm60 yes need your vehicle schematics and need some easy to navigate way to it

rusefillc, 2022-01-21:
@shadowm60 question about 

https://github.com/rusefi/rusefi/wiki/Vault-Of-Mercedes-OEM#m111---i4

new link caption is "OEM Wiring harness" which implies that it's for universal OEM Wiring harness for ANY M111

I believe that's not the case? shall we add some important details into "OEM Wiring harness" text like year? mention EUDM maybe?

duhafnusa4, 2022-02-28:
the pnp is for pre evo motors (2002 or older) and you can use this picture which shows the differences. 

[image]

duhafnusa4, 2022-02-28:
I would add this picture to help differentiate Pre and Post evo engines (our pnp ecu is for pre) 

[image]

rusefillc, 2022-02-28:
@shadowm60 ^^^ golden content!

mi-hol, 2022-12-11:
@shadowm60 is there anything left to be done or should ir be closed?

shadowm60, 2022-12-11:
To be honest i did everythink i could come up with, i think we can close and open new ticket if we have direct req for something?

closed 2022-12-11 by rusefillc