rusefi/rusefi_documentation #201: What's our goal with google search results?

rusefillc, 2022-01-22:
At the moment https://www.google.com/search?q=HOWTO+M73+v12+on+Proteus returns https://rusefi.com/docs/wiki/HOWTO/HOWTO-M73-v12-on-Proteus/ which properly redirects to https://wiki.rusefi.com/HOWTO/HOWTO-M73-v12-on-Proteus/

Do we expect this to change say in a week or two?

rusefillc, 2022-01-22:
See #189

chuckwagoncomputing, 2022-01-23:
Yes this should change.

rusefillc, 2022-02-03:
Not yet.

chuckwagoncomputing, 2022-02-05:
Maybe you should [request a recrawl](https://developers.google.com/search/docs/advanced/crawling/ask-google-to-recrawl)
You could submit the sitemap which is at https://wiki.rusefi.com/sitemap.xml
Also see the Index Status report and URL Inspection tool linked from that Google document.

chuckwagoncomputing, 2022-02-19:
Looks like it's showing up on DuckDuckGo, but not on Google. @rusefillc have you done anything from my last comment yet?

rusefillc, 2022-02-19:
[image]

rusefillc, 2022-02-19:
I was planning to request the recrawl and totally forgot to do so. Doing it now because it's totally the right thing to do.

rusefillc, 2022-02-19:
[image]

rusefillc, 2022-02-19:
[image]

rusefillc, 2022-02-19:
Note "Dec 31, 2021" not sure if that's the global "Last crawled"

rusefillc, 2022-02-19:
[image]

[image]

chuckwagoncomputing, 2022-02-20:
Looks good. I guess we wait a few days/weeks and see if it shows up.

rusefillc, 2022-03-01:
things got worse? no wiki result returned any more?

chuckwagoncomputing, 2022-03-01:
Looks like the status of 347 pages is:
Discovered - currently not indexed: The page was found by Google, but not crawled yet. Typically, Google wanted to crawl the URL but this was expected to overload the site; therefore Google rescheduled the crawl. This is why the last crawl date is empty on the report.

and 14 are
[Crawled](https://support.google.com/webmasters/answer/7643010) - currently not indexed: The page was crawled by Google, but not indexed. It may or may not be indexed in the future; no need to resubmit this URL for crawling.

48 pages are fully indexed.
So I guess we wait some more?

chuckwagoncomputing, 2022-03-16:
We're up to 65 fully indexed. Disappointing.
Also, I noticed that Google corrects "HOWTO" to "HOW TO", and then that particular page doesn't show up, but if you click on "Search instead for HOWTO M73 v12 on Proteus", it's the first result. Frustrating.

rusefillc, 2022-09-20:
good news:
https://www.google.com/search?q=rusefi+dfu+button

includes https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwjnxvb25KP6AhVxAjQIHXtQAUMQFnoECAwQAQ&url=https%3A%2F%2Fwiki.rusefi.com%2FHOWTO%2FHOWTO-Update-Firmware%2F&usg=AOvVaw3o6K1GoEhFSutRYHb3jEHx

chuckwagoncomputing, 2022-09-21:
Where is the wiki hosted? It's on 1&1 right? Is it possible that you don't have enough bandwidth for Google to crawl without overloading your hosting?
As a reminder, this is the official explanation from Google:

Discovered - currently not indexed

The page was found by Google, but not crawled yet. Typically, Google wanted to crawl the URL but this was expected to overload the site; therefore Google rescheduled the crawl. This is why the last crawl date is empty on the report.

mck1117, 2022-09-22:
how does google even know that it would be expected to overload you?

chuckwagoncomputing, 2022-09-22:
I assume it measures the download rate while indexing a few pages.

mi-hol, 2022-12-11:
redirects to https://wiki.rusefi.com/HOWTO/HOWTO-M73-v12-on-Proteus/ now, therefore closing

closed 2022-12-11 by mi-hol

rusefillc, 2022-12-11:
🥳 🎈 @chuckwagoncomputing we did it! YOU did it!!!