rusefi/rusefi_documentation #203: Is new wiki ready for prime time?

rusefillc, 2022-01-23:
Are there any reason NOT to switch rusefi.com main page links to new wiki?

for instance https://wiki.rusefi.com/Technical/All-Supported-Triggers/ instead of https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers etc etc

chuckwagoncomputing, 2022-01-23:
In my mind #188 is a pretty big issue.

chuckwagoncomputing, 2022-01-25:
Also, lots of pages are missing titles at the top, which is where MkDocs gets its page names.
e.g. https://wiki.rusefi.com/Vaults/Vault-Of-Honda-OEM/ is called "OBD-1" in the search results.

rusefillc, 2022-01-25:
Sounds like we need to start documenting wiki conventions which are consequences of mkdocs

Just a quick checklist

Realistically we might need automation to validate or even implement "first line of file matches file name" for instance

chuckwagoncomputing, 2022-01-25:
Pages that need to be updated to reflect wiki3:
Build-Server-and-Automation.md
Documentation-Strategy.md
Documentation-Workflow.md
HOWTO/HOWTO-contribute-to-documentation.md

chuckwagoncomputing, 2022-01-26:
#205 When we're ready

chuckwagoncomputing, 2022-02-19:
What are we waiting on at this point?

rusefillc, 2022-02-20:
@chuckwagoncomputing I do not know :( I am ready when you are ready, I think we've done enough just to start switching.

https://github.com/rusefi/rusefi_documentation/issues/208 was the last nail in https://github.com/rusefi/rusefi/wiki value now that we've lost in-place edits https://wiki.rusefi.com/ is clearly in the lead

See also https://github.com/rusefi/rusefi_documentation/issues/210

duhafnusa4, 2022-04-27:
been a few months. How is the new wiki?

rusefillc, 2022-11-08:
Open question if #245 is a blocker or not

mi-hol, 2022-11-13:
> Open question if #245 is a blocker or not

not from my view because "search to provide results in optimal order" is a complete different unrelated topic.
I'd suggest to close this issue.

chuckwagoncomputing, 2023-01-07:
I vote that rusefi.com links be switched to wiki.rusefi.com

mi-hol, 2023-03-24:
> I vote that rusefi.com links be switched to wiki.rusefi.com

so do I and I would create a PR

are the files below the only places to require change or do I miss something?
[image]

[image]

chuckwagoncomputing, 2023-03-24:
Those need to be switched to links that work in either wiki.
I was talking about links in other places, e.g. TS help links, firmware readmes, the forum and Slack.

https://github.com/search?q=repo%3Arusefi%2Frusefi%20%22github.com%2Frusefi%2Frusefi%2Fwiki%22&type=code

Also, I see that I misspoke myself. What I meant was:
I vote that github.com/rusefi/rusefi/wiki links be switched to wiki.rusefi.com