rusefi/rusefi_documentation #207: Some rusefi.com/s links are out of date

chuckwagoncomputing, 2022-02-05:
```
[david@archlinux rusefi]$ grep -r "rusefi.com/s/" firmware/tunerstudio/rusefi.input 
                field = "!https://rusefi.com/s/trigger"
            field = "!https://rusefi.com/s/vvt"
                field = "!https://rusefi.com/s/fuel"
                field = "!https://rusefi.com/s/debugmode"
                field = "#See rusefi.com/s/lua for more info"
        webHelp = "https://rusefi.com/s/fuel"
        webHelp = "https://rusefi.com/s/fuel"
        webHelp = "https://rusefi.com/s/etb"
        webHelp = "https://rusefi.com/s/tpstps"
                field = "https://rusefi.com/s/etb"
```

Some of these link to Wiki1 pages, some link to Wiki2 pages, and some link to Wiki2 pages that don't exist.

rusefillc, 2022-02-05:
We have an issue with https://github.com/rusefi/web_backend/issues/146?

chuckwagoncomputing, 2022-02-06:
Why not point to wiki3?

rusefillc, 2022-02-06:
I am happy to use wiki3, just a bit lost if we are using wiki3 or not yet

mi-hol, 2022-12-11:
@rusefillc I understood in our "intro session" that wiki3 is active, hence can this be closed?

rusefillc, 2022-12-11:
#167 says that wiki2-human is still primary

rusefillc, 2022-12-12:
@mi-hol ?

mi-hol, 2022-12-12:
how can I help?

rusefillc, 2022-12-12:
I am confused. What is the open problem statement? I am not aware f anything being broken

mi-hol, 2022-12-12:
> What is the open problem statement?

Just needed time to digest answer.

closed 2022-12-12 by mi-hol