rusefi/rusefi_documentation #213: rusefi.com landing page Bounce rate too high

rusefillc, 2022-03-19:
Morgan brings up poor Bounce rate metric at https://rusefi.com/

Our Bounce rate* is pretty high... At 50-60%
(*Users who go to the site and don't click on any links)

[image]

pk386, 2022-03-19:
suggestion remove bottom 1/3 of top banner.

Current: 
[image]

Proposed: 
[image]

mi-hol, 2022-11-19:
With changes applied in March, how did  Bounce rate metrics develop over time?
From my view navigation at bottom of page is a non-standard approach, hence may contribute to the high bounce rate

rusefillc, 2022-11-20:
No idea what this means just pasting the report screenshot

@chris-tina FYI

[image]

rusefillc, 2022-11-20:
Comparison but important to note  that we did ads

[image]

mi-hol, 2022-12-11:
@rusefillc seems unclear what actions this issue should trigger. 
Would suggest to prioritize as low or even close.
What are your thoughts?

rusefillc, 2022-12-11:
I do not know what actions could be performed

With my overall focus not-on-retail this is overall low priority for me, but I am happy to facilitate if there are specific low budget actions anyone has in mind

mi-hol, 2022-12-11:
thanks, I created and assigned labels to reflect your statement

mi-hol, 2023-01-12:
@chuckwagoncomputing would you have visibility of Google Analytics for this site?

chuckwagoncomputing, 2023-01-12:
Yes I do.
I'm not a web designer and I have no idea what could help this.

MashaTreyster, 2026-11-03:
i see that a lot of users come to forum for 0 sec - look like bots, so they produce useless traffic.
Users that came for shopping and products read significant amount of time.
[image]