rusefi/rusefi_documentation #217: users are confused between three ways to log data

rusefillc, 2022-06-20:
https://rusefi.com/forum/viewtopic.php?p=45629#p45629

rusefillc, 2022-06-20:
:( rusEFI console https://github.com/rusefi/rusefi/tree/master/java_console which has unconditional constant logging

rusefillc, 2022-06-20:
what pages are linking new https://github.com/rusefi/rusefi/wiki/Logging-Guide?

does TS .ini help link to https://github.com/rusefi/rusefi/wiki/Logging-Guide in any way? maybe tooltip URLs or else?

mck1117, 2022-06-20:
> what pages are linking new https://github.com/rusefi/rusefi/wiki/Logging-Guide?

[image]

I've added it to the sidebar

rusefillc, 2022-06-20:
we need a link from SD card dialog in TS

chuckwagoncomputing, 2025-12-17:
I think this has been addressed.