rusefi/rusefi_documentation #220: MRE GND inconsistency

rusefillc, 2022-06-24:
https://user-images.githubusercontent.com/48498823/175553641-20bb13db-6037-44d8-bd94-2a7c8602fb7a.mp4

rusefillc, 2022-06-24:
https://user-images.githubusercontent.com/48498823/175554297-99ab3799-4dac-4c89-9a1d-c695b343e46e.mp4

rusefillc, 2022-06-24:
hopefully progress?

[image]

rusefillc, 2022-06-24:
More issues #221

chuckwagoncomputing, 2022-06-24:
> Q: what is the source of static pinout PNG? I do not remember :( Best I can do is drop it and rely only on interactive pinout. Relying only on interactive could be a bad idea I believe we have complaints about poor static rendering of interactive pinout?

What do you mean by "static rendering" and what are the complaints? If they can be improved I will do what I can.

rusefillc, 2022-06-24:
#222 created for static rendering

rusefillc, 2022-12-11:
i am the owner

too low on the priority list for me to look into this :(