rusefi/rusefi_documentation #222: text right on pins mode

rusefillc, 2022-06-24:
can we have a mode where text for all pins goes right above pins photo?

i.e. instead of 
[image]

can we have closer to 

[image]

as for XY problem - sometimes one pic/table is more efficient

no idea how to implement in terms of mode switch or not, in terms of image scale or not

chuckwagoncomputing, 2022-06-24:
I might be able to figure something out. One thing that will make it difficult is that we don't have a "YAML ground truth" for that kind of label. The 'type' field isn't specific enough, and the 'function' field is too long.

rusefillc, 2022-06-24:
overall we lack clear consistent best practice

* some boards append pin numbers manually
* overall TS vs function separation is unclear

we either need 3rd text version or a clean-up :)

chuckwagoncomputing, 2022-07-01:
Suppose the label in TS were "<pin number> - <the text from the above MRE table>"
Would that be sufficient information for the end user to have in front of them in TS, or is there currently important information in the current TS Name field that the user would loose?

If it would be enough, I propose we make the TS Name field 4 or 5 letters max, and combine the pin number and TS name when generating the INI.

mi-hol, 2022-12-11:
@rusefillc a decision seems required

chuckwagoncomputing, 2023-04-28:
What do you think of this as a solution?

## Cons
  - Requires hovering
  - Won't work on mobile

## Pros
  - Doesn't require restricting the length of the info text
  - Doesn't require making the text smaller to fit

https://user-images.githubusercontent.com/1531127/235257980-08d37d3a-edd9-4aed-ab58-9b6ce9d1dece.mp4

mi-hol, 2023-04-29:
> * Won't work on mobile

Could you please elaborate on reason(s) for this?
From my view this is the main issue preventing acceptance

chuckwagoncomputing, 2023-04-29:
I was thinking that you couldn't hover on mobile, but it does show up if you tap on a pin.
[image]

chuckwagoncomputing, 2023-04-29:
I have added this to the latest release of interactive-pinout. It's an optional feature that requires adding an `info-column` parameter in the workflow.

@rusefillc If you like this, I can PR an updated workflow file to use this.

rusefillc, 2023-04-29:
@chuckwagoncomputing yes please PR!

rusefillc, 2023-04-29:
https://rusefi.com/docs/pinouts/microrusefi/ looks pretty cool thank you @chuckwagoncomputing

closed 2023-04-29 by rusefillc