rusefi/rusefi_documentation #225: Hellen NB2 link takes you to wiki landing page

duhafnusa4, 2022-06-28:
@BeerMoneyMotorsports When I click on the NB2 link (https://github.com/rusefi/rusefi/wiki/Hellen72 shown in image below) it links to the rusefi wiki landing page. 

[image]

Where is the NB2 Hellen page?

duhafnusa4, 2022-06-28:
OK I found it here.

https://github.com/rusefi/rusefi/wiki/Hellen-NB2

Can I fix link and change it to https://github.com/rusefi/rusefi/wiki/Hellen-NB2 ?

rusefillc, 2022-06-28:
@duhafnusa4 yes please attempt to fix link accorind to docs workflow

duhafnusa4, 2022-06-28:
Where is the rev F schematic and iBom?

duhafnusa4, 2022-06-28:
This is the connector correct?

[image]

BeerMoneyMotorsports, 2022-06-28:
That is the connector.  
[hellen72_NB2-f-schematic.pdf](https://github.com/rusefi/rusefi_documentation/files/9004486/hellen72_NB2-f-schematic.pdf)

rusefillc, 2022-06-28:
https://github.com/rusefi/rusefi/wiki/Hellen-One-Platform link fixed

rusefillc, 2022-06-28:
NB2-F added to https://github.com/rusefi/rusefi/wiki/Hellen-NB2

duhafnusa4, 2022-06-28:
Why on Miata hellen do we call some by year and some by chassis?

I suggest we change it to
Miata NA
Miata NB1
Miata NB2

[image]

BeerMoneyMotorsports, 2022-06-29:
the 90-97 NA chassis is split between two ecus.  Unfortunately it is  not a clean split.  There are enough pinout differences, that we cannot use one ecu for all of them.  I would certainly love a cleaner naming method.

90-95 cover NA6 and NA8
96/97 cover NA8 with OBD

BeerMoneyMotorsports, 2022-06-29:
That looks backwards depending on the view you are using.  see pic below is the back of the connecter where the wires insert.

[image]

mi-hol, 2022-12-11:
@rusefillc how to deal with the new issue in [comment](https://github.com/rusefi/rusefi_documentation/issues/225#issuecomment-1169428849) ?
I'd recommend to open a new issue as the issue from title seems solved.

rusefillc, 2022-12-11:
@BeerMoneyMotorsports uses his own tracking and priorities, closing

closed 2022-12-11 by rusefillc