rusefi/rusefi_documentation #226: Detail on how to set up iDrive or equivalent CAN button control

duhafnusa4, 2022-06-29:
I have an iDrive controller I want to use for some functions (boost control, launch control, cruise etc...) 

I found the CAN data from danilo081 on https://www.bimmerfest.com/threads/k-can2-messages-to-wake-a-cic-idrive-controller.890419/#post-13182279 :

My controler 9317695
K-CAN2

Can bus Standard 11bit identifier - 500kbps [for K-CAN 100kbps ]

Pinout
1 +12 1 |*|
2 -GND 2 |*|
3 CAN HIGH 3 |*|
4 CAN LOW 4 |* |

# WAKE UP
###############################
-> 202: FD
-> 202: FD
//send it twice to wake up controler

-> 563: 63 // start sending this every 1000-1500 milisec to keep it active

# INIT
##################################
//On first reply <- "5E7: 99 01 1D 16 06 FF FF FF" respond with

-> 273: 1d e1 00 f0 ff 7f de xx

//8th byte xx takes values between 00 and FF
//I got success not sending 8th byte at all but 100% working solution is to send 4th recieved byte from 5E7 message above -> [16]
//5th byte from 5E7 = [06] - not initialized [01]-initialized
// now buttons and rotary are initialized and next 5E7: message have 5th byte = 01

# BACKLIGHT
##################################
//To regulate backlignt send

-> 202: YY

//YY is from 00 tp FD to dim
//"202 FE" wil turn light off and if you stop sending "563: 63" iDrive cotroller will go to sleep

# SLEEP
#################################
//to put controler to sleep again, stop sending -> "563: 63" and send

-> 202: FE

# BUTTON DEFINITIONS
################################
// fife butons have c0 on 5th byte
// 6th byte is button specific
// 4th byte is event specifyc

<- 267: E1 FD YY XX C0 08 AUDIO
<- 267: E1 FD YY XX C0 01 MENU
<- 267: E1 FD YY XX C0 40 TEL
<- 267: E1 FD YY XX C0 04 OPTIONS
<- 267: E1 FD YY XX C0 02 BACK

// XX - 4th byte takes values:
// 01 - press
// 02 - hold
// 00 - relese
// YY is just counter of messages goning in cycle from 00-ff and around again allways increasing

# Joystick press and directions
#################################
// Stick button have DE on 5th byte
// Direction buttons have DD on 5th byte Stick button jabe DE on 5th byte
// 4th byte is buton and event specific

## STICK-BUTTON
<- 267: e1 fd 3b 01 de 01 - press
<- 267: e1 fd 3b 02 de 01 - hold
<- 267: e1 fd 3c 00 de 01 - relese

## STICK-UP
<- 267: e1 fd 3d 11 dd 01 - press
<- 267: e1 fd 3d 12 dd 01 - hold
<- 267: e1 fd 3e 00 dd 01 - relese

## STICK-RIGHT
<- 267: e1 fd 43 21 dd 01 - press
<- 267: e1 fd 43 22 dd 01 - hold
<- 267: e1 fd 44 00 dd 01 - relese

## STICK-DOWN
<- 267: e1 fd 3f 41 dd 01 - press
<- 267: e1 fd 3f 42 dd 01 - hold
<- 267: e1 fd 40 00 dd 01 - relese

## STICK-LEFT
<- 267: e1 fd 41 81 dd 01 - press
<- 267: e1 fd 41 82 dd 01 - hold
<- 267: e1 fd 42 00 dd 01 - relese

# Joystick ROTATION
#################################
// Bbte 1 and 2 are fixed
// Byte 3 is just counter of messages goning in cycle from 00-ff and around again, allways increasing
// Byte 4 turn positions, 00-ff and around again increesing for CW and decreasing for CCW by [+-1]
// Byte 5 "round counter" when byte 4 is FF and with CW one more "CLICK" increase byte 5 by one and oposite when byte 4 is 00 one more clik in CCW direction byte 5 going -1
// my solution on code is to calculate turns directions is to "JOIN" Byte5 and Byte4 "B5B4"bp8 = [b5]*0x100 + [b4] and calculate difference betwen turns
// EXAMPLE:
// we are at <- "264: e1 fd 4c ff 1f 1e"
// [0x1F]*0x100+[0xFF] = 1FFF(hex) = 8191(dec) CW turn adds +1
// so we get <- "264: e1 fd 4d 00 20 1e"
// [0x20]*0x100+[0x00] = is 2000(hex) = 8192(dec)

## CW turns examples
<- 264: e1 fd 4c ff 1f 1e
<- 264: e1 fd 4d 00 20 1e
<- 264: e1 fd 4e 01 20 1e
<- 264: e1 fd 4f 02 20 1e
<- 264: e1 fd 50 03 20 1e
<- 264: e1 fd 51 04 20 1e

## CCW turns turns examples
<- 264: e1 fd 52 03 20 1e
<- 264: e1 fd 53 02 20 1e
<- 264: e1 fd 54 01 20 1e
<- 264: e1 fd 55 00 20 1e
<- 264: e1 fd 56 ff 1f 1e
<- 264: e1 fd 57 fe 1f 1e
<- 264: e1 fd 58 fd 1f 1e

# responses
#############
<- 277: E1 9D 04 FF // just reply on init, for me not of interest
<- 567: 00 00 00 00 00 00 00 67 // If you see this you are late on "563: 63" message so reduce delay

NOTE: When controller is at sleep, send first message 2 or 3 times. I am sending 202 fd twice to wake it up. You will see lights.

I am gathering information here and when I have it all I will create information on GH wiki for how to set it up using LUA.

rusefillc, 2022-06-30:
https://rusefi.com/forum/viewtopic.php?f=4&t=2325

https://github.com/rusefi/rusefi/wiki/Lua-Scripting#bmw-idrive

mi-hol, 2022-12-11:
Status of issue seems unclear. Who is supposed to take action(s)?