rusefi/rusefi_documentation #227: Document 12 tooth trigger

rusefillc, 2022-07-04:
https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers should explain usages of 

[image]

rusefillc, 2022-07-04:
See also https://rusefi.com/forum/viewtopic.php?p=45801&sid=79166fc0cd1805031193243a839e6322#p45801

rusefillc, 2022-10-16:
@mck1117 please increase priority of this

See also #241

mi-hol, 2022-12-11:
todo:  on https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers
verify below text for relevance
```
See 'trigger decoding' in [Doxygen](http://rusefi.com/docs/html/)

May, 2020: Subaru SVX added

April, 2020: Honda K 12+1 added.

April, 2020: Renix 44-2-2 added.

Dev note: unit_tests executable produces triggers.txt file - gen_trigger_images.bat reads triggers.txt and produces these .png files. TODO: automate this further?
```

mi-hol, 2023-01-12:
@mck1117 Hi Matt, may I kindly remind about this open action?

closed 2023-01-12 by rusefillc