rusefi/rusefi_documentation #229: Wall Wetting AE Link Empty

moistbrochure, 2022-07-08:
As per the request of Sir @duhafnusa4 the wall wetting Ae info page is not there. 
[image]
https://github.com/rusefi/rusefi/wiki/wall_wetting

However, we have this page https://github.com/rusefi/rusefi/wiki/X-tau-Wall-Wetting which is full of stuff but not full of the other AE settings/tuning besides the x-tau stuff.

duhafnusa4, 2022-07-08:
Where did you find the link https://github.com/rusefi/rusefi/wiki/wall_wetting ?

moistbrochure, 2022-07-08:
Overview tab
https://github.com/rusefi/rusefi/wiki/overview

rusefillc, 2022-07-08:
https://github.com/rusefi/rusefi/wiki/Overview fixed

moistbrochure, 2022-07-08:
Looks like the "about rusefi" link on the forum takes you to an overview page that has been forgotten about. 
[image]
It takes you to https://github.com/rusefi/rusefi/wiki/Overview which has the link to the wall-wetting page, that is empty. Not the good wall wetting description page. 

The wiki main page https://github.com/rusefi/rusefi/wiki does not include the place to download latest firmware and what not, but it does have the correct wall wetting link.

moistbrochure, 2022-07-08:
Rodger that. New ticket added #230

rusefillc, 2022-07-08:
Now I am getting it, we have two different issues here

#231 which is my duplicate of #230 

and fact that we have two clones of same page

rusefillc, 2022-07-08:
@chuckwagoncomputing do you know anything about https://github.com/rusefi/rusefi/wiki/Overview and https://github.com/rusefi/rusefi/wiki content duplication? I assume we shall wipe out Overview ? replace with link to home?

chuckwagoncomputing, 2022-07-08:
I don't remember anything about it. I say we replace it with a link to home.

rusefillc, 2022-07-08:
Done https://github.com/rusefi/rusefi/wiki/Overview

closed 2022-07-08 by rusefillc