rusefi/rusefi_documentation #235: MRE GP clarification

duhafnusa4, 2022-09-19:
pinout shows "GP" for the pins, but in the description below it there are 3 versions (GP LOW/HIGH/PP)

the pinout diagram needs clarification on which GP is which, unless they all could be any 3 (and that needs to be detailed)

[image]

[image]

rusefillc, 2022-09-19:
This seems to be about https://github.com/rusefi/rusefi/wiki/Hardware-microRusEFI-wiring

Source files are lost of some or all of images on that page, the way forward would be to replace everything with interactive pinout

Which brings us to https://github.com/rusefi/rusefi_documentation/issues/236

duhafnusa4, 2022-09-22:
whats the timeline for #236?

rusefillc, 2022-10-11:
@duhafnusa4 does https://github.com/rusefi/rusefi_documentation/issues/236 show closed to you?

[image]

rusefillc, 2022-10-11:
next step is to decide what exact text to change in what exact way on https://rusefi.com/docs/pinouts/microrusefi/

rusefillc, 2022-10-11:
final step would be to drop outdated images and table from https://github.com/rusefi/rusefi/wiki/Hardware-microRusEFI-wiring once https://rusefi.com/docs/pinouts/microrusefi/ has the best info

mi-hol, 2022-12-11:
> next step is to decide what exact text to change in what exact way

@rusefillc  guess you would own decision making or can it be delegated to someone else?