rusefi/rusefi_documentation #239: PPS reading crazy values while voltages and scaling is correct

ElDominio, 2022-10-07:
https://rusefi.com/online/view.php?log=1067

Raw voltages seem fine yet PPS bounces all over the place?

ElDominio, 2022-10-07:
[image]

ElDominio, 2022-10-07:
Solved myself, needed to recalibrate; why does this happen?

rusefillc, 2022-10-07:
@ElDominio i really hope that @mck1117 would ask for more info

like before calibration tune
after calibration tune
details on how did you get the before-calibration tune :)

ElDominio, 2022-10-07:
same as always, grab up grab down

both tunes are on rusefi online

mi-hol, 2022-12-11:
@rusefillc it seems this issue is in the wrong repository as it seems TunerStudio related

rusefillc, 2022-12-11:
> both tunes are on rusefi online

@ElDominio two months later is it still trivial to locate those?